luc-github/ESP3D #633: [BUG]telnet service does not work with proper telnet client

pmjdebruijn, 2021-05-14:
**Describe the bug**
When I connect with the telnet client to the ESP3D telnet service Marlin does not accept any sent commands.
I suspect newline handling to be at the heart of the problem.

**To Reproduce Failure**
```
telnet esp3d
M114
echo:Unknown command: ""
```

**To Reproduce Success**
```
nc esp3d 23
M114
X:0.00 Y:-6.00 Z:0.00 E:0.00 Count X:0 Y:-960 Z:0
```

**Expected behavior**
I would expect at least for the proper telnet client to work, and optionally nc as well.

**ESP3D Firmware:**
- Version: 3.0 branch (aaca73fe568291b0988653e5d6fd92a92aa56f9f)
- Wifi mode: STA
- Flash method: [SKR E3 RRF bootloader]

**Target Firmware:**
 - Name: Marlin
 - Version: 2.0.8

**Board used:**
 - MCU: ESP07s (ESP8266)
 - Name: SKR E3 RRF onboard module
 - Flash size: 4M

**Additional context**
- Telnet sends `M114\r\n`
- Netcat sends `M114\n`

luc-github, 2021-05-14:
did you tried putty?

pmjdebruijn, 2021-05-14:
I'm not sure why that would matter? As I said, netcat works fine as a workaround.

The fact that PuTTY apparently needs to be put into raw mode, sortof confirms my issue I guess...

luc-github, 2021-05-14:
yes what is supported is raw tcp , what is used by pronterface and repetier host, it not a "true" telnet.

pmjdebruijn, 2021-05-14:
Ok, fair enough, but I'd suggest not calling it telnet (in code/documentation), and using a different port than 23, as both those things heavily imply telnet should work.

luc-github, 2021-05-14:
pr and help are always welcome

luc-github, 2021-05-15:
@pmjdebruijn does your telnet client work with https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino and Marlin ?

closed 2021-06-16 by stale[bot]