luc-github/ESP3D #638: [FEATURE REQUEST] support for BTT SKR2 + esp8266

LawnMo, 2021-05-21:
**Is your feature request related to a problem? Please describe.**
Hey, I recently bought BTT's new board and it appeared, according to the documentation, that their "wifi RRF extension module" was usable under marlin too. There's a wifi / U-disk jumper, that afaik let us select which is connected to spi. It clearly states "when U-disk jumpers are on, wifi module only works in uart and only marlin can be used".  
My first thoughts was to plug that wifi module and flash esp3d (2.1.1) on it then plug it back in (after modifying marlin's serial_port attributions, pin file says SERIAL_PORT or SERIAL_PORT2 needs to be 3 for esp to work).  
Except, so far nothing seems to work, esp3d receives marlin's serial messages (sdcard remove/inserted) but it can't send commands (anything M117, M22, G28..).  

**Describe the solution you'd like**
I checked the solders on both the motherboard and the wifi module, there's nothing shorted, everything is clean and there's continuity between the two (so it's not a faulty pin header either).  
So I moved on to pin definitions, which is where i noticed something "familiar", BTT SKR2's wifi module is an esp12s and the breakout board expose 2x 8pins, just like the MKS robin Lite.  
Upon further inspection :  
[image]

It's using GPIO0 & 4 and as far as I can tell, esp3d 2.1.1 on a esp8266 uses GPIO0 & 2, which would explain why I get messages from marlin but can't send any commands (?)

Which is why I'm wondering, should I try to use esp3d 3.x and build it with :  
`#define COMMUNICATION_PROTOCOL MKS_SERIAL`  
?

**Describe alternatives you've considered**
This could also be a configuration error, a hardware error, a documentation error, I'm kinda lost and this hardware is not yet properly documented ^^".

**Additional context**
Hope I can have somebody's opinion/solution on that semi-issue. Thanks for reading this :)

luc-github, 2021-05-21:
under Marlin only the normal serial mode is available not the SPI,  be sure the serial pins are  not used for somethig else and not shared with USB under Marlin. 

The Serial pins on ESP are GPIO 1 and GPIO 3 (PD8/PD9), GPIO 0 and GPIO 4 are used for the SPIO protocol of RRF which also use SPI.

The MKS protocol is for MKS boards not BTT, and Marlin support only MKS protocol for MKS boards so far. Fast upload could be implemented only if https://github.com/luc-github/ESP3D/issues/575 is implemented - which is far from be done at this moment

LawnMo, 2021-05-21:
> 
> The Serial pins on ESP are GPIO 1 and GPIO 3 (PD8/PD9), GPIO 0 and GPIO 4 are used for the SPIO protocol of RRF which also use SPI.

Oh that's useful, thanks. I can't really tell where the leads go to on the board (multi layers and only the silkscreen is available) but I kinda assumed the jumpers were there to route a couple SPI pins to one or the other (wifi VS udisk).  
Now I guess I mostly need to match the RX/TX pins on my esp for marlin to properly communicate, thanks for the clarification :)

closed 2021-05-21 by LawnMo

luc-github, 2021-05-21:
@LawnMo please share your findings and how you flashed the ESP board, this is very interresting
I already got some requests about this board and capabilities - but I do not have any, so your experience is more than valuable

LawnMo, 2021-05-21:
Will do if I can get anything working ^^.  
Flashing the BTT RRF Wifi module was pretty straight forward, a quick examination shows the esp12s is just soldered directly to a breakout board, no crossed pins, no additional components (no resistors, no voltage regulator on the back) :  
[image]  
(source: https://www.sensatio.io/tutorial-how-to-prepare-your-esp8266-esp-12-for-flashing).  
I wired RX/TX and GND from the ESP to an usb-to-serial dongle and got the chip in flash mode as detailed on the schematics (grounding GPIO0 and RST).  

On the marlin side, it only needs PORT_SERIAL 3 (or PS2 or PS3), so far, esp3d gets serial messages but can't send, so either a pin conflict or a faulty board, I guess.

LawnMo, 2021-05-31:
workaround found : https://github.com/MarlinFirmware/Marlin/issues/21982  

tl,dr: BTT SKR2 rev A & B have a small pin conflict, esp3d works "natively" on this board 👍

luc-github, 2021-05-31:
great ^_^ thank you for sharing