luc-github/ESP3D #642: [FEATURE REQUEST] Question about DEFAULT_IP_VALUE

trengtor, 2021-06-15:
As I understand it, DEFAULT_IP_VALUE describes the IP address for the gateway in AP mode.
But in network client mode, this is the client's address.
How to specify the gateway separate IP address for the network client mode?

luc-github, 2021-06-15:
Do not change DEFAULT_IP_VALUE it is a default value for setup 
use webUI or direct commands to change IP https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands

trengtor, 2021-06-15:
> Do not change DEFAULT_IP_VALUE it is a default value for setup
> use webUI or direct commands to change IP https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands

I use DEFAULT_IP_VALUE as an static IP-addres for ESP in LAN client mode.

luc-github, 2021-06-16:
I move this thread to discussions as it is not an issue

closed 2021-06-16 by luc-github