luc-github/ESP3D #650: Why is my esp3d only working on the TTL adapter?

niekjuuhh, 2021-07-15:
Hi reader,

I was wondering if someone could help me out with the following problem:

I am running esp3d on my esp8266-01s chip. The problem is that the chip only starts the wifi when its connected with the ttl adapter. When i plug exactly the same chip on my skr 1.4 turbo the esp8266 is not starting the signal.

Please help me out

Regrads

luc-github, 2021-07-15:
is your ESP01S from BTT or from another supplier?

niekjuuhh, 2021-07-17:
> is your ESP01S from BTT or from another supplier?
> looks same as #598

Thanks luc for your reply, Thats working fine now. The next problem i have is that the printer is not communicating with the webserver.

Hardwareserial.h already has been set to the right configurations because thats a standard of platformio now.

Do i need to add #define LPC_PINCFG_UART3_P4_28 somewere? Inside the configuration.h file it isn't working.

Please help me out.

luc-github, 2021-07-17:
this pure marlin configuration which Change quite often, did you checked https://youtu.be/NGgzw-XayEo or Marlin github issue tracker ?

luc-github, 2021-07-17:
be sure baudrate are identical in Marlin and ESP3D and serial port configuration is same as video

niekjuuhh, 2021-07-17:
> this pure marlin configuration which Change quite often, did you checked https://youtu.be/NGgzw-XayEo or Marlin github issue tracker ?

I followed the video, changed everything like him. But for some reason the printer is not connected to the webserver :(

niekjuuhh, 2021-07-17:
> be sure baudrate are identical in Marlin and ESP3D and serial port configuration is same as video

Both are 250000!

luc-github, 2021-07-17:
share your configuration.h of Marlin to see if something pop up - but I cannot be sure I can help

closed 2021-08-15 by luc-github