luc-github/ESP3D #656: Reset all settings - BTT E3 RRF V1.1

toreandreas, 2021-07-30:
Hello.
I have managed to make a mistake. I have changed the password on WIFI in webui, and now I can no longer enter. Both in AP and in client mode. This is because I was unable to connect to my network. I think #define MAX_PASSWORD_LENGTH was set to 16 (which is too low). And then I have appostrophe in the password, which I know has been problematic before. Can I change something in firmware that deletes / resets webui, or is there another solution?
The card is as mentioned in the heading a BIGTREETECH E3 RRF V1.1. Software was downloaded here on 28.07.2021.
Thank you in advance.

luc-github, 2021-07-30:
What Firmware version are you using ? 2x or 3.0 ?
the command to reset setting is `[ESP444]RESET`

toreandreas, 2021-07-30:
ESP3D-2.1.x (https://github.com/luc-github/ESP3D). 
I followed https://www.youtube.com/watch?v=7J9vFdrF9kQ, but used higher baud rate, 250,000.
I'm new to 3D printing, and even newer in ESP3D... Where to enter the command [ESP444] RESET? (The terminal for octopi?). Does anything need to be changed in any firmware first?

luc-github, 2021-07-30:
well on 2.1 it won't work, because even for serial it will ask for admin password - the only solution is to erase full flash , then reflash

luc-github, 2021-07-30:
Ok I just watched the video - I did not know the bootlader supported ESP3D.BIN file on SD to flash the ESP
So my suggestion :
1 - disable Authentication in ESP3D then flash like you did
2 - modify the admin and user passwords
3 - flash again by enabling authentication

luc-github, 2021-07-30:
I have a doubt which password did you changed ?

toreandreas, 2021-07-30:
1 - Will this stop the ESP3D from entering webui?

I changed password in webui, for AP mode, and Client mode.

I am not at home now, so I can not test yet.

luc-github, 2021-07-30:
ok the mention of 

>#define MAX_PASSWORD_LENGTH was set to 16

confused me, so you do not enabled authentication - right ?

In that case you can send any serial commands to reset all or just setup wifi : https://raw.githubusercontent.com/wiki/luc-github/ESP3D/docs/Commands2_1.txt

to do this just use the serial of your printer and use the `M118 P0`

but as I am not sure Marlin will forward answer from ESP3D serial, so better to do full reset
```
M118 P0 [ESP444]RESET
M118 P0 [ESP444]RESTART
```
But after looking at video, if you followed it, seems the default values are changed in code, so before, you need to put back the originales value for password : 12345678 , like this you will only modify in webui and a reset will always allow you to have a working AP

toreandreas, 2021-07-30:
First, it was not the #define #define MAX_PASSWORD_LENGTH, it was MAX_LOCAL_PASSWORD_LENGTH. So there was nothing wrong, except my memory.
-
Flashed with ESP3D.bin (Without any changes in config.h, or other files.)

Tried [ESP444] RESET in octopi / terminal:
_Send: [ESP444]RESET
Recv: echo:Unknown command: "[ESP444]RESET"
Recv: ok_
Tried logging in with password 12345678, no success.

Tried M118 P0 [ESP444]RESET, and M118 P0 [ESP444]RESTART (also in octopi/terminal):
_Send: M118 P0 [ESP444]RESET
Recv: [ESP444]RESET
Recv: ok
[...]
Send: M118 P0 [ESP444]RESTART
Recv: [ESP444]RESTART
Recv: ok_
Tried logging in with password 12345678, Success. 
I have to disconnect from the internet with this PC, but will be back.

toreandreas, 2021-07-30:
Success confirmed.
I think M118 P0 [ESP444] RESET, and M118 P0 [ESP444] RESTART were the fixed .

Everything works, but not apostrophe in the WIFI password. I noticed one thing. About my password is: abcdef'ghijklmn. I wrote this in the wizard. When I later checked up on this it just said: abcdef. It is possible that I use is not an apostrophe, but another sign.
I have no idea where I can check this out, here may not be the right place to ask.
Anyway, thanks for all the help.

toreandreas, 2021-07-30:
Is it possible to enter (or otherwise) enter passwords like Hexadecimal? 
See http://jorisvr.nl/wpapsk.html

luc-github, 2021-07-30:
only passphrase is supported 
the password of 32 chars has in theory no limitation - 
if it has it can be :
1 - a limitation of esp core (e.g.: before space where not working, it was fixed after a while)
2 - a bug in webUI with some char is not properly encoded - to solve this I need:
a - a confirmation that M118 P0 [ESP106]xxx'xxxx does not work and allow you to connect
b - a clear version of what password is not working
Note: password is not displayed in theory once key in- so I do not understand you can see it

toreandreas, 2021-07-30:
a - M118 P0 [ESP106]xxx'xxxx do not work.
 Should it be M118 P0 [ESP106] xxx'xxxx. Anyway, no connection.
b - The password consists of the following: uppercase and lowercase letters, space and an apostrophe. 25 characters. First character is a space.
Have also tested with xxx\'xxxx, without success.

-I have useed some big letters, above....

toreandreas, 2021-07-30:
A slash (or whatever it's called) disappeared. Should be xxxslash'xxxx.

toreandreas, 2021-07-30:
I think the symbol I have used is not an apostrophe, but a singel quote. '

closed 2021-08-15 by luc-github