luc-github/ESP3D #660: ESP3D 3.0 Out of order.

AntoszHUN, 2021-08-01:
ESP3D V3.0 fails VSC compilation. He writes an error all the time.

[image]
[image]

luc-github, 2021-08-01:
it looks like a PlatformIO - check you platformIO installation, also be sure you have git installed - I saw some problem with PIO when git is not installed under windows and PIO not always request the installation

AntoszHUN, 2021-08-01:
> 
> 
> it looks like a PlatformIO issue - check you platformIO installation, also be sure you have git installed - I saw some problem with PIO when git is not installed under windows and PIO not always request the installation

PlatformIO IDE is installed. But what is a git?

luc-github, 2021-08-01:
https://git-scm.com/download/win

AntoszHUN, 2021-08-01:
Another mistake.
[image]

luc-github, 2021-08-01:
just tested and it is ok on my side
[image]

be sure you do a clean before building

AntoszHUN, 2021-08-01:
> 
> 
> just tested and it is ok on my side
> [image]
> 
> be sure you do a clean before building and of course use ESP32Cam env

esp32dev ran flawlessly. But esp32cam doesn't want to be good.

luc-github, 2021-08-01:
what are 
`[ESP800]`
and 
`[ESP420]plain` 

output ?

AntoszHUN, 2021-08-01:
I had to rewrite the board. board = esp32cam is now good. : D

AntoszHUN, 2021-08-01:
Good. I just can't see the wifi network.

luc-github, 2021-08-01:
[image]

it is a gif so quality is not great but real ui is ok 
[image]

luc-github, 2021-08-01:
One bug is new : in AP mode wifi scan is not working anymore ..., it was before...
 I need to check if API changed in latest ESP32 core - or if it is a bug in core

AntoszHUN, 2021-08-01:
> 
> 
> One bug is new : in AP mode wifi scan is not working anymore ..., it was before...
> I need to check if API changed in latest ESP32 core - or if it is a bug in core
> EDIT: fixed...my mistake, one line was missing...

All right. I download and test.

AntoszHUN, 2021-08-01:
I can't upload to it.
[image]

luc-github, 2021-08-01:
you changed the partition size - so Filesystem is corrupted: you do not have 1GB = 1023MB of Flash
you need to format 
open terminal tab by clicking it and type `[ESP710]FORMAT` then refresh

AntoszHUN, 2021-08-01:
> 
> 
> you changed the partition size - so Filesystem is corrupted: you do not have 1GB = 1023MB of Flash
> you need to format
> open terminal tab by clicking it and type `[ESP710]FORMAT` then refresh

And I didn't rewrite anything.

AntoszHUN, 2021-08-01:
I formatted and reinstalled. I get the same thing in the picture.

luc-github, 2021-08-01:
you do not need to reinstall, just Format using esp command and eventualy restart board but refresh should enough, after flash size should be some kb

AntoszHUN, 2021-08-01:
> 
> 
> you do not need to reinstall, just Format using esp command and eventualy restart board but refresh should enough, after flash size should be some kb
> share ui fs size after proceed

You don't want to format me with the ESP command.

luc-github, 2021-08-01:
1 Type [ESP710]FORMAT
2 Refresh , if size is still incorrect Restart board

AntoszHUN, 2021-08-01:
> 1 Type [ESP710]FORMAT
> 2 Refresh , if size is still incorrect Restart board

Where could i write it?

luc-github, 2021-08-01:
like I told you before

[image]

AntoszHUN, 2021-08-01:
> 
> 
> like I told you before
> 
> [image]

Oh my stupid. Thanks! It worked. What files do I need to do now? just so I don't spoil it.

luc-github, 2021-08-01:
[image]
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/dist/printer/production/index.html.gz

closed 2021-08-01 by AntoszHUN