luc-github/ESP3D #663: [BUG] blank configuration page after uploading 404.html file

digant73, 2021-08-12:
I received from BTT the ESP-01S with the "esp3d.bin" fw already flashed (so no need to use BTT writer to flash the fw).
I plugged the esp-01s on my BTT TFT35 V3 and it was properly detected reporting the IP 192.168.0.1 on the TFT display.
From the browser I connected to 192.68.0.1 and I uploaded "index.html.gz" and "favicon.htm". Then I tried to upload "404.htm" but the upload freezed at 62% while the available space reported was 92%. Maybe the available space was not enough for "404.htm".
Now I'm not able to complete the setup. If I try to connect to 192.168.0.1, no page is displayed (only a white page).
Is there any possibility to restore the setup page without the need to reflash the fw on esp-01s (I don't have the BTT writer nodule)?

Attached the files used

[ESP-01S.zip](https://github.com/luc-github/ESP3D/files/6977480/ESP-01S.zip)

luc-github, 2021-08-12:
connect your esp using embedded webui

http://192.168.0.1?forcefallback=yes

you should be able to delete files and upload new one, only index.html.gz is mandatory others are not

digant73, 2021-08-12:
yes I was able to delete the files. I uploaded only index.html.gz but the problem is still there. I cannot complete the setup pressing on the refresh button (in browser or setup page). I used the files from BTT repo.
Is there the latest binary available in your github?

luc-github, 2021-08-12:
I do not do any binary, may be there is incompatibility between btt binary and the index.html.gz you are trying to use, I suggest you to use the one used by btt if any
I have no idea what is the problem with info you give.
why did you uploaded index.html.gz? the preloaded version was not working?

digant73, 2021-08-12:
problem fixed. The BTT file was probably corrupted. I downloaded the index.html.gz from https://github.com/luc-github/ESP3D-WEBUI, uploaded and then moved to 192.168.0.1 to complete the setup. From http://192.168.0.1?forcefallback=yes it is not possible to complete the setup. Thanks for the help! nice work

luc-github, 2021-08-12:
ho great, so issue is solved. 
can we close it ?

digant73, 2021-08-12:
sure

closed 2021-08-12 by digant73

digant73, 2021-08-13:
I just had some time to use/test your FW and WEBUI. It is very good,
Just some small suggestions for a further WEBUI release:
1) In the ESP3D Settings page, when setting the "Wifi mode" I would also disable (or use different colors) all the settings not related to the selected wifi mode. So, if "client station" is selected, I would disable all the "AP" settings. Or, if you want to allow to set AP just as the failover solution in case the esp3d is not able to connect to the client station at startup, I would consider to use different colors just to understand which is the primary and secondary access choice
2) In the Dashboard page, I would make more easy to use the print panel. Once a file is slected and the print button is pressed, I will always provide a status bar (e.g. below the file list) providing the file beeing printed with the pause/resume/stop buttons. Also the files list should be made disabled just to avoid to select and print another file while already printing a file. On BTT TFT fw I could eventually make the changes in order to handle the pause/resume/stop gcode commands in case not yet properly supported.

luc-github, 2021-08-13:
The ESP3D 3.0 is totaly different than the 2.X
1 - the settings are more splitted to avoid confusion, also I do not want to disable AP settings when  STA is selected because you should be able to set your AP information as it is the default mode when connection failed
2 - the print status  is already implemented in 3.0 - not the play/pause buttons yet
check webUI status (I am already refactoring the 3.0 as I did not do well...):
https://github.com/luc-github/ESP3D-WEBUI/discussions/94
https://github.com/luc-github/ESP3D-WEBUI/issues/157
https://github.com/luc-github/ESP3D-WEBUI/issues/144

digant73, 2021-08-13:
Yes I supposed you didn't disable AP settings just to have a valid failover (default) in case STA failed.
Ok I will upload the WEBUI 3.0 and test it. Is there any reason/issue (maybe due to BTT TFT fw) for the pause/resume buttons not yet present?

luc-github, 2021-08-13:
The WebUI 3.0 need ESP3D 3.0
The rewrite is not yet usable - better stick to 3.0
Unfortunaly you habe a ESP01 which mean not enough space to do the web update, you need to use a usb-serial adapter to flash the esp board.
That said if you just want to have a look to WebUI 3.0 or 3.0rewrite you can use the offline version - it does not need an ESP Board and run on any Computer simulating ESP3D answers

digant73, 2021-08-13:
ok, I will buy the adapter. Is the BTT Writer ok or do you suggest a better adapter?
For WEBUI branch 3.0, I suppose the dist files to be used are in dist/printer/production, right?

luc-github, 2021-08-13:
btt writer is fine for esp01
yes locaton is correct