luc-github/ESP3D #667: [Question] Connecting ESP32 to the MKS TFT24

primus192, 2021-09-10:
Hi. I have got printer with MKS GEN L board. My printer have MKS TFT24 touchscreen connected to AUX1 port. As i am concerned, this is the only usable serial port on the board and if i wanted to use esp3d i need to disconnect the touchscreen... However my TFT24 touchscreen have WiFi port on the back. Can i connect my ESP32 Devkit v1 to this WiFi port and somehow route the esp3d to mainboard via this connector? Can you give me any advice how to do this? Thanks!

luc-github, 2021-09-10:
you cannot use same serial port for esp board and TFT, both are doing queries and the data flow willl be merged inconsistantly
Connecting ESP32 Devkit v1 to TFT WIFI port will work for moniitoring but unlikly for file upload - the file upload need super clean signal due to usage of very high baud rate, and with esp 32 devkit connected by cable will will have some EMI noise for sure that will corrupt high baud rate transfert

primus192, 2021-09-10:
Okay, so what type of connection this issue covers? #523

luc-github, 2021-09-10:
MKS WIFI devices connected with MKS WIFI module

primus192, 2021-09-10:
So i cant upload esp3d on the module shown on the photo? 
[image]

luc-github, 2021-09-10:
this one is not an esp32 dev kit, it is MKS WIFI module, so it works

primus192, 2021-09-10:
Ok i see it is esp8266, so my esp32 should work if i correctly connect it. Will give it a try, thanks.

luc-github, 2021-09-10:
no that is my point - put esp3d on it is not the issue
file upload / transfer may not work due to your connection having emi noise, but it may also work if you are lucky

primus192, 2021-09-10:
I will try and will send you feedback/screenshots, thanks :)

primus192, 2021-09-10:
This is needed to uncomment it?

//MKS_SERIAL Mks protocol

luc-github, 2021-09-10:
no this the value to put as reference
read this : https://github.com/luc-github/ESP3D/discussions/620

primus192, 2021-09-10:
Got a problem with uploading webui, it says file upload failed(13)

luc-github, 2021-09-10:
It means size error 
can you do a screenshot of page with error ?

primus192, 2021-09-10:
[image]

luc-github, 2021-09-10:
I think you FS is corrupted  - you should not have 95.60KB used space if nothing on FS
go to Terminal tab and click on it
type `[ESP710]Format` then refresh page and try upload again

primus192, 2021-09-10:
[image]

primus192, 2021-09-10:
[image]
I tried to push this file manually via VSCODE to the esp32 it works but when i click od ESP3D white screen appears and nothing is shown

luc-github, 2021-09-10:
I am confused what board did you flashed ?

primus192, 2021-09-10:
Esp32 Devkit V1

luc-github, 2021-09-10:
what env did you selected in VS CODE ?

primus192, 2021-09-10:
I just manually uploaded index.html.gz to the spiffs, because i couldnt do it via 192.168.0.1

luc-github, 2021-09-10:
This is not my question - what env did you selected in VS Code for flashing your ESP32 ?

primus192, 2021-09-10:
I flashed my esp32 via Arduino IDE

luc-github, 2021-09-10:
what settings did you used  to flash in arduino then

primus192, 2021-09-10:
Ok i selected wrong board, it is working now

primus192, 2021-09-10:
[image]
I need to connect it to the printer in order to see control panel?

luc-github, 2021-09-10:
You need to define the target FW - it is repetier in your case, and also you need to enable TFT SD and TFT USB in settings

primus192, 2021-09-10:
[image]
Its working.. Thanks ;)

luc-github, 2021-09-11:
Good I move this to discussion area then

closed 2021-09-11 by luc-github