luc-github/ESP3D #669: I can't find/connect to a printer

ManoelVictor9, 2021-09-16:
I'm trying for days connecting my printer in esp3d, however, on the esp3d local website, I can't find or connect the printer in printer configurations, I tried changing the serial port, but nothing changed.
I'm using an esp8266 (NodeMcu) and MKS SGEN_L V2.0 Board in my printer.

luc-github, 2021-09-17:
please check the template as bot say
you do not provide any information to allow to allow to help you

ManoelVictor9, 2021-09-17:
> please check the template as bot say
> you do not provide any information to allow to allow to help you

Done, I put more informations and pictures.

luc-github, 2021-09-17:
Ok here what come to mind that need to check
* is only Printer configuration that does not respond or all commands ?
if not only here more check:
* Did you set proper port com for ESP board in Marlin configuration?
* do you use same baud rate between Marlin and ESP
* are you sure you did not put Rx->Rx and Tx->Tx instead of Rx->Tx and Tx-.Rx ?

*if only printer config is not working 
* what is the M503 output ?

ManoelVictor9, 2021-09-17:
> Ok here what come to mind that need to check
>
> * is only Printer configuration that does not respond or all commands ?
>   if not only here more check:
> * Did you set proper port com for ESP board in [...]

I think the connections are plugged-in property and the Marlin configuration too, but I can't get any information from the printer sensors or use any command on the ESP3D website to the MKS board, so the MKS board and esp8266 are working separately, so I realized that the problem is in the transmition, but the question is where
That's the configuration of esp in platformio
[image]

luc-github, 2021-09-17:
you do not need to change anything in platformIO if you use the latest git version - the git version has everything setup for esp8266
https://github.com/luc-github/ESP3D/blob/2.1.x/platformio.ini#L39-L64

ManoelVictor9, 2021-09-17:
> you do not need to change anything in platformIO if you use the latest git version - the git version has everything setup for esp8266
> https://github.com/luc-github/ESP3D/blob/2.1.x/platformio.ini#L39-L64

Thanks, however I tried with this configuration and It didn't work. :/ I'll Join in ESP3D Discord with the developers and try to figure out why this is happening.

luc-github, 2021-09-17:
developers?
I suggest you the check  I have  listed

DivingDuck, 2021-09-18:
... and even better he should check what pins he try to connect.

luc-github, 2021-09-26:
@ManoelVictor9 Any update?