luc-github/ESP3D #670: [FEATURE REQUEST]Direct SD upload, I see that you have written a part.

xiaoyaoking, 2021-09-19:
Direct SD upload, I see that you have written a part. Do you have a plan to complete this function?

ESP8266 I test that I can read and write. It's messy to write my own code. I still hope I can use what you wrote.

My English is not good. I hope you can understand it.

[image]

luc-github, 2021-09-19:
Direct SD upload is implemented in 3.0 so I do not understand your feature request

xiaoyaoking, 2021-09-19:
> Direct SD upload is implemented in 3.0 so I do not understand your feature request

i 'm sorry... I don't see 3.0. I'm using 2.1

closed 2021-09-19 by xiaoyaoking

luc-github, 2021-09-19:
https://github.com/luc-github/ESP3D/tree/3.0