luc-github/ESP3D #672: [FEATURE REQUEST]<Enter comprehensive title>

DvoraNoob, 2021-09-22:
I don't know if it has compatibility, but could I add a connectivity option with a tuya socket?

luc-github, 2021-09-23:
tuya xompatibility? do you have link to the modele you are refering to?

DvoraNoob, 2021-09-23:
> tuya xompatibility? do you have link to the modele you are refering to?

Like this https://plugins.octoprint.org/plugins/tuyasmartplug/ with any smart switch
I don't know if is possible.

Legsmaniac, 2021-09-23:
Easy to add your own with V3.0 by creating your own custom panel. 
I've done this for my Sonoff switch controlled via Home Assistant.

DvoraNoob, 2021-09-23:
> Easy to add your own with V3.0 by creating your own custom panel.
> I've done this for my Sonoff switch controlled via Home Assistant.

I tried, but doesn't work. 
I don't know what ip I put in the box