luc-github/ESP3D #675: [BUG]<index.html.gz won't get flashed with platformio. Uploading to AP fails, too>

jwagn96, 2021-09-26:
**Describe the bug**
Interface for file&firmware uploads, asking me for the missing index.html.gz, that should usually be uploaded already. Uploading manually results in fails, so that there's no chance for me to get any further. The upload stops repeatedly at exact 73%. With every try, some space of the file system remains occupied (ca. 3kB)
**To Reproduce**
Steps to reproduce the behavior:
1. Upload firmware with Platformio
2. Upload file system
3. Access ESP3D AP
4. Try uploading index.html.gz

**Screenshots**
[image]
[image]

**ESP3D Firmware:**
- ESP core version: 2.5.2
- Version:2.1.1.b8
- Web UI Version:[e.g 2.1b14]  
- Flash method:  PlatformIO

**Board used (please complete the following information):**
 - MCU: ESP8266 
 - Name: ESP-01
 - Flash size: 1m

luc-github, 2021-09-26:
Why did you changed the settings provided in PlatformIO?

 * you do not use the esp8266 core 2.5.2 but more recent one that is not tested
 * Using 80Mhz is not good idea, use 160Mhz to get proper response from MCU
 * The platformio settings I provided are for 1M ESP8266 so you do not need to change anything
Please use the tested settings before trying any changes,

I also suggest you to erase all flash first, the menu is available in platformIO,  as your FileSystem may be corrupted

https://github.com/luc-github/ESP3D/blob/2.1.x/platformio.ini
```
[env:esp01s_160mhz-2.7.4]
platform = espressif8266@2.6.2
board = esp01_1m
framework = arduino
monitor_speed = 115200
monitor_flags = --echo
monitor_filters = colorize
board_build.f_cpu = 160000000L
board_build.f_flash = 40000000L
board_build.flash_mode = dout
board_build.ldscript = eagle.flash.1m256.ld
upload_resetmethod = nodemcu
build_flags =
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore =
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD

[env:esp01s_160mhz-2.5.2]
platform = espressif8266@2.2.0
platform_packages = toolchain-xtensa@2.40802.200502
board = esp01_1m
framework = arduino
monitor_speed = 115200
monitor_flags = --echo
monitor_filters = colorize
board_build.f_cpu = 160000000L
board_build.f_flash = 40000000L
board_build.flash_mode = dout
board_build.ldscript = eagle.flash.1m256.ld
upload_resetmethod = nodemcu
build_flags =
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore =
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD 
```

luc-github, 2021-09-28:
@jwagn96 any update?

jwagn96, 2021-09-29:
Yes, today I finally got to test your suggestion. It works wonderfully just by using your code in the platformio.ini instead of mine. There were no additional changes needed. Thank you for your kind help! 
Using ArcWelder for post processing, the G-Code of parts is transmitted in no time, even with the slow serial connection between ESP and mainboard.

closed 2021-09-29 by jwagn96