luc-github/ESP3D #676: [BUG] Upload Rejected(5) - Error (5) File Transfert (sic) Failed

gtr33m, 2021-10-01:
I'm unable to update the firmware of Branch 3.0 using the main web interface or the fallback interface.
[image]

[image]
Firmware: 3.0.0.a95  
Interface: 3.0.0.86 
 
Browser version: Firefox 78
Chip ID: 6205
CPU Frequency: 160 Mhz
Free memory: 20.27 KB
SDK: 2.2.2-dev(38a443e)
Flash Size: 4.00 MB
Available Size for update: 475.44 KB
Filesystem type: LittleFS
Filesystem usage: 680.00 KB/2.98 MB
Baud rate: 115200
Sleep mode: None
WiFi: Enabled
Hostname: TTB
HTTP port: 80
Telnet port: 23
WebDav port: 8181
Client mode: Enabled
Mac address: F4:CF:A2:63:18:3D
SSID: Hawkins
Signal: 82 %
Phy Mode: 11n
Channel: 4
IP Mode: DHCP
IP: 192.168.50.104
Gateway: 192.168.50.1
Mask: 255.255.255.0
DNS: 192.168.50.1
Access point mode: Disabled
Mac address: F6:CF:A2:63:18:3D
Serial: Enabled
Notifications: ON(telegram)
Buzzer: Disabled
Serial: MKS
Target firmware: Marlin
FW architecture: ESP82XX

**ESP3D Firmware:**
- ESP core version: 3.0.0
- Version: a95
- Web UI Version: 3.0.0.86 
- Wifi mode: STA  
- Flash method: PlatformIO
- set to MKS_SERIAL
 
**Target Firmware:**
 - Name: Marlin
 - Version 2.0

**Board used (please complete the following information):**
 - MCU: ESP82XX
 - Name:MKS Wifi 
 - Flash size: 4M
 - [env:esp01s_160mhz]

**Additional context**
I can upload the firmware file to the filesystem as with the index.html.gz so if there is a way to apply it from there, this might solve the problem.

I have formatted the flash using `[ESP710]FORMAT` which does return a confirmation in the terminal, but I'm not getting anything from the board itself and the webui does not show any data, so my current firmware is not working as intended.

luc-github, 2021-10-01:
why did you set 3MB for FS ? there is no space for the FW update - the report show
`Filesystem usage: 680.00 KB/2.98 MB`
`Available Size for update: 475.44 KB`
I recommand you to change the flash size to 2MB FS 1MB OTA

gtr33m, 2021-10-11:
Sorry for the late reply, I didn't get any notification.

I followed the recommendation from MKS [https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/Use-MKS_Robin_WIFI-on-Robin-Nano#use-esp3d-firmware-on-mks-robin-wifi](url)

Board: Generic ESP8266 Module
Flash Mode: DOUT
Flash Frequency: 40Mhz
Flash Size: 4M (3M SPIFFS)
CPU Frequency: 160Mhz Debug Port: Disabled
Debug Level: None
Reset Method: CK
Upload Speed: 115200

I've managed to compile using either platform.io and arduino ide, and can re-compile with the 2MB FS 1MB OTA, but the question is, how do I flash it. Is there any way other than physically flashing the esp12s? I don't have a nodemcu or programming board, and all the instructions I can find seem to call for a 10k resistor, which I also don't have.

Is it possible to downgrade over wifi to a version which is smaller than 475.44 KB and then re-flash with a new flash size?

Legsmaniac, 2021-10-19:
> ...... but the board I have is a Leonardo......
> ...... I could have ordered a new module for next to nothing and just started over, but I'm too stubborn!

I would kindly suggest you swallow your pride and make life easier for yourself.  I fear you may still end up ordering a better module because, yes, the Leo is exactly as you've read, often more problematic than it's worth.  Likewise, some cheap Chinese boards can give you plenty headaches too.  I speak as a long time user of many different Arduino boards over the years and have several (and I mean several) boards dotted about the home all running various IoT functions 24/7 without a hitch.  That is not before losing my hair over several problematic boards which have ended up in the bin!
Good luck.

luc-github, 2021-11-05:
Any update ? Can we close issue ?

closed 2021-11-08 by luc-github