luc-github/ESP3D #677: [BUG] compilation via PlatformIO fails with CAMERA_DEVICE defined

tom2199, 2021-10-03:
**Describe the bug**
If trying to compile via PlatformIO (5.2.0) using the current commit 248ce51c6f38383f233dc1de031fbc09171c4e51 and an uncommented `#define CAMERA_DEVICE CAMERA_MODEL_AI_THINKER` in https://github.com/luc-github/ESP3D/blob/248ce51c6f38383f233dc1de031fbc09171c4e51/esp3d/configuration.h#L219 results in an error shown here:

<details>
  <summary>Log</summary>

```
platformio run -e esp32cam
Processing esp32cam (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
['C:\\Users\\Tom\\Documents\\PlatformIO\\Projects\\ESP3D\\libraries', '$PROJECT_LIBDEPS_DIR\\$PIOENV', 'C:\\Users\\Tom\\.platformio\\lib']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.bd54ee4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa-esp32 8.4.0+2021r1
Converting esp3d.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
|-- <EEPROM> 2.0.0
|-- <Preferences> 2.0.0
|-- <BluetoothSerial> 2.0.0
|-- <Ticker> 2.0.0
|-- <Wire> 2.0.0
|-- <lv_arduino> 2.0.3
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <TFT_eSPI> 1.4.11
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <FFat> 2.0.0
|   |-- <FS> 2.0.0
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <FS> 2.0.0
|-- <SD> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <LittleFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <SD_MMC> 2.0.0
|   |-- <FS> 2.0.0
|-- <ESP32SSDP> 1.1.1
|   |-- <WiFi> 2.0.0
|-- <Update> 2.0.0
|-- <ESP8266-Arduino-Lua> 0.0.30
|-- <ArduinoOTA> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ESPmDNS> 2.0.0
[... 595 lines cut ...]
esptool.py v3.1
Merged 1 ELF section
============================================ [SUCCESS] Took 138.46 seconds ============================================

Environment    Status    Duration
-------------  --------  ------------
esp32cam       SUCCESS   00:02:18.461
============================================= 1 succeeded in 00:02:18.461 =============================================
```
</details>

In this commit, PlatformIO is able to compile a build with `CAMERA_DEVICE CAMERA_MODEL_AI_THINKER` defined.

**To Reproduce**
Steps to reproduce the behavior:
1. clone 3.0 branch
2. edit configuration.h in line https://github.com/luc-github/ESP3D/blob/248ce51c6f38383f233dc1de031fbc09171c4e51/esp3d/configuration.h#L219 (uncomment it)
3. run command ` platformio run -e esp32cam`
4. See error

**Expected behavior**
Sucessfully compile the build using the `#define CAMERA_DEVICE CAMERA_MODEL_AI_THINKER` for the enviroment `esp32cam`

**Board used (please complete the following information):**
 - MCU: ESP32-CAM

luc-github, 2021-10-03:
I am not sure where the issue come from 
on my side :
```
esp32cam       SUCCESS   00:01:31.767
================================================================= 1 succeeded in 00:01:31.767 =================================================================

Terminal will be reused by tasks, press any key to close it.

> Executing task: platformio run --environment esp32cam <

Processing esp32cam (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
['/home/luc/Documents/GitHub/luc-github/ESP3D/libraries', '$PROJECT_LIBDEPS_DIR/$PIOENV', '/home/luc/.platformio/lib']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.317be68 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa-esp32 8.4.0+2021r1
Converting esp3d.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
|-- <EEPROM> 2.0.0
|-- <Preferences> 2.0.0
|-- <BluetoothSerial> 2.0.0
|-- <Ticker> 2.0.0
|-- <Wire> 2.0.0
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <TFT_eSPI> 1.4.11
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <lv_arduino> 2.0.3
|-- <FFat> 2.0.0
|   |-- <FS> 2.0.0
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <FS> 2.0.0
|-- <SD> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <LittleFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <ESP32SSDP> 1.1.1
|   |-- <WiFi> 2.0.0
|-- <Update> 2.0.0
|-- <ESP8266-Arduino-Lua> 0.0.30
|-- <ArduinoOTA> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ESPmDNS> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <HTTPClient> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <WiFiClientSecure> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <BMx280MI> 1.2.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <DHT sensor library for ESPx> 1.0.6
|-- <WebSockets> 2.3.5
|   |-- <WiFi> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0
|   |   |-- <WiFi> 2.0.0
Building in release mode
Compiling .pioenvs/esp32cam/src/esp3d.ino.cpp.o
Retrieving maximum program size .pioenvs/esp32cam/firmware.elf
Checking size .pioenvs/esp32cam/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.5% (used 57504 bytes from 327680 bytes)
Flash: [======    ]  64.4% (used 1266405 bytes from 1966080 bytes)
================================================================= [SUCCESS] Took 9.45 seconds =================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32cam       SUCCESS   00:00:09.446
================================================================= 1 succeeded in 00:00:09.446 =================================================================

Terminal will be reused by tasks, press any key to close it.
```

luc-github, 2021-10-03:
I use VSCode - I do not use command line -  may be there is a difference ?

tom2199, 2021-10-03:
In the output you provided I cant see this items logged, maybe the problem stems from this? VSCode provides the same output.
```
|-- <SD_MMC> 2.0.0
|   |-- <FS> 2.0.0
```

luc-github, 2021-10-03:
[image]

[image]

I have removed all esp package and it download the one screened

I just enabled the camera from configuration.h not the SD

tom2199, 2021-10-03:
Here is my output from the newest commit (with comment in place) within VSCode:
```
> Executing task: C:\Users\Tom\.platformio\penv\Scripts\platformio.exe run --environment esp32cam <

Processing esp32cam (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: esp32dev; 
framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
['C:\\Users\\Tom\\Documents\\PlatformIO\\Projects\\ESP3D\\libraries', '$PROJECT_LIBDEPS_DIR\\$PIOENV', 'C:\\Users\\Tom\\.platformio\\lib']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.bd54ee4 
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa-esp32 8.4.0+2021r1
Converting esp3d.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
|-- <EEPROM> 2.0.0
|-- <Preferences> 2.0.0
|-- <BluetoothSerial> 2.0.0
|-- <Ticker> 2.0.0
|-- <Wire> 2.0.0
|-- <lv_arduino> 2.0.3
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <TFT_eSPI> 1.4.11
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <FFat> 2.0.0
|   |-- <FS> 2.0.0
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <FS> 2.0.0
|-- <SD> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <LittleFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <SD_MMC> 2.0.0
|   |-- <FS> 2.0.0
|-- <ESP32SSDP> 1.1.1
|   |-- <WiFi> 2.0.0
|-- <Update> 2.0.0
|-- <ESP8266-Arduino-Lua> 0.0.30
|-- <ArduinoOTA> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ESPmDNS> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
[... 427 lines cut ...]
Compiling .pioenvs\esp32cam\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pioenvs\esp32cam\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pioenvs\esp32cam\FrameworkArduino\libb64\cdecode.c.o
Compiling .pioenvs\esp32cam\FrameworkArduino\libb64\cencode.c.o
Compiling .pioenvs\esp32cam\FrameworkArduino\main.cpp.o
Compiling .pioenvs\esp32cam\FrameworkArduino\stdlib_noniso.c.o
Compiling .pioenvs\esp32cam\FrameworkArduino\wiring_pulse.c.o
Compiling .pioenvs\esp32cam\FrameworkArduino\wiring_shift.c.o
Archiving .pioenvs\esp32cam\libFrameworkArduino.a
Linking .pioenvs\esp32cam\firmware.elf
Retrieving maximum program size .pioenvs\esp32cam\firmware.elf
Checking size .pioenvs\esp32cam\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.2% (used 46372 bytes from 327680 bytes)   
Flash: [======    ]  60.4% (used 1188481 bytes from 1966080 bytes)
Building .pioenvs\esp32cam\firmware.bin
esptool.py v3.1
Merged 25 ELF sections
================================================= [SUCCESS] Took 194.37 seconds =================================================
Environment    Status    Duration
-------------  --------  ------------
esp32cam       SUCCESS   00:03:14.368
================================================== 1 succeeded in 00:03:14.368 ==================================================
Terminal will be reused by tasks, press any key to close it.
```

tom2199, 2021-10-03:
Here is the same output just with diffent Line 219:
```
> Executing task: C:\Users\Tom\.platformio\penv\Scripts\platformio.exe run --environment esp32cam <

Processing esp32cam (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; board: esp32dev; 
framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r1
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r1
['C:\\Users\\Tom\\Documents\\PlatformIO\\Projects\\ESP3D\\libraries', '$PROJECT_LIBDEPS_DIR\\$PIOENV', 'C:\\Users\\Tom\\.platformio\\lib']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.bd54ee4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa-esp32 8.4.0+2021r1
Converting esp3d.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WiFi> 2.0.0
|-- <EEPROM> 2.0.0
|-- <Preferences> 2.0.0
|-- <BluetoothSerial> 2.0.0
|-- <Ticker> 2.0.0
|-- <Wire> 2.0.0
|-- <lv_arduino> 2.0.3
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <TFT_eSPI> 1.4.11
|   |-- <SPIFFS> 2.0.0
|   |   |-- <FS> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <FFat> 2.0.0
|   |-- <FS> 2.0.0
|-- <SPIFFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <FS> 2.0.0
|-- <SD> 2.0.0
|   |-- <FS> 2.0.0
|   |-- <SPI> 2.0.0
|-- <LittleFS> 2.0.0
|   |-- <FS> 2.0.0
|-- <SD_MMC> 2.0.0
|   |-- <FS> 2.0.0
|-- <ESP32SSDP> 1.1.1
|   |-- <WiFi> 2.0.0
|-- <Update> 2.0.0
|-- <ESP8266-Arduino-Lua> 0.0.30
|-- <ArduinoOTA> 2.0.0
|   |-- <Update> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <ESPmDNS> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <ESPmDNS> 2.0.0
[... 96 lines cut ...]
Compiling .pioenvs\esp32cam\src\src\modules\ethernet\ethconfig.cpp.o
Compiling .pioenvs\esp32cam\src\src\modules\filesystem\esp_filesystem.cpp.o
Compiling .pioenvs\esp32cam\src\src\modules\filesystem\esp_globalFS.cpp.o
esp3d/src/modules/camera/camera.cpp:27:10: fatal error: fd_forward.h: No such file or directory

********************************************************************
* Looking for fd_forward.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:fd_forward.h"
* Web  > https://platformio.org/lib/search?query=header:fd_forward.h
*
********************************************************************

 #include "fd_forward.h"
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\esp32cam\src\src\modules\camera\camera.cpp.o] Error 1
================================================== [FAILED] Took 50.28 seconds ==================================================
Environment    Status    Duration
-------------  --------  ------------
esp32cam       FAILED    00:00:50.278
============================================= 1 failed, 0 succeeded in 00:00:50.278 =============================================The terminal process "C:\Users\Tom\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32cam'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

luc-github, 2021-10-03:
Ho boy - seems uninstalling package is not enough, I just delete all cache and now can reproduce the camera issue...,
seems a package issue...

luc-github, 2021-10-03:
The camera code changed and I missed it - I need to see how to put back the missing code

luc-github, 2021-10-03:
Should be fixed with https://github.com/luc-github/ESP3D/commit/e903dce63f78966f154a9dbc7b424c12085886e7

tom2199, 2021-10-03:
Thank you, it works on my end as well!

closed 2021-10-03 by tom2199