luc-github/ESP3D #678: [Question]<How to set up GPIO for Ethernet>

liu2-3zhi, 2021-10-03:
I want to configure Ethernet, but I found that the PINs of MDC, MDIO, and CLK_IN need to be set. I rummaged through all the documents and did not find the answer.

luc-github, 2021-10-03:
On which board ?

liu2-3zhi, 2021-10-03:
I want to know the default GPIO port, and then draw the board by myself.

luc-github, 2021-10-03:
which mcu ?

liu2-3zhi, 2021-10-03:
LAN8720A-CP-TR

luc-github, 2021-10-03:
ESP32 ?  - ESP 8266 ?

liu2-3zhi, 2021-10-03:
esp32

luc-github, 2021-10-03:
I have only tested the olimex :
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware
https://github.com/espressif/arduino-esp32/blob/master/variants/esp32-poe/pins_arduino.h

liu2-3zhi, 2021-10-03:
So, which file is the GPIO port set in? I did not find pins_arduino.h at https://github.com/luc-github/ESP3D.

luc-github, 2021-10-03:
because I just used the board as it is , same as sample code :
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720/ETH_LAN8720.ino

luc-github, 2021-10-03:
https://github.com/OLIMEX/ESP32-POE/blob/master/HARDWARE/ESP32-PoE-hardware-revision-D/ESP32-PoE_Rev_D.pdf
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/ETH.h

If you need let me know - I did not added any config in pins.h as I did not needed

liu2-3zhi, 2021-10-04:
Thanks, I understand, it should be connected like this
GPIO23-MDC

GPIO18-MDIO

GPIO17 CLK_IN

GPIO21-TX_EN

GPIO19-TX0

GPIO22 TX1

GPIO25-RX0

GPIO26-RX1

GPIO27-CRS_DRV

liu2-3zhi, 2021-10-04:
But I still recommend adding relevant configuration to pins.h

closed 2021-10-04 by liu2-3zhi

luc-github, 2021-10-04:
PR is welcome - you are the one who can check it easily

liu2-3zhi, 2021-10-04:
> PR is welcome - you are the one who can check it easily

sorry, English is'n the official language in my country,so I poor with English.I can't understand what you say.

luc-github, 2021-10-04:
what is your language and country ?

liu2-3zhi, 2021-10-04:
> what is your language and country ?

Chinese ,China
I'm just a junior high school student.

liu2-3zhi, 2021-10-04:
I first started contacting ESP3D just because I wanted to DIY a 3D printer of my own

luc-github, 2021-10-04:
没问题，你可以 push code to github, 因为你可以很容易看看行不行

liu2-3zhi, 2021-10-04:
Unfortunately, I still don't know much about editing and modifying C+ for the time being.

luc-github, 2021-10-04:
学习,实践,没有其他的办法
加油

liu2-3zhi, 2021-10-04:
> 学习,实践,没有其他的办法 加油

Can I add you to Telegram?

luc-github, 2021-10-04:
I do not use telegram, I just setup for notification setup, you can join esp3d discord server https://discord.gg/95fWFMEn