luc-github/ESP3D #690: [BUG]<Enter comprehensive title>

Anion-anion, 2021-11-08:
luc Hi!
I am used ESP3d v 3.0 with SD card. 
 - MCU: ESP32
 - Name: ESP32 Devkit v4 
 - Flash size: 4M

I connected the sd reader to the esp32 to contacts
CS  D5 
SCLK D17 
MISO D19 
MOSI D23 

the configuration of the ESP3D is as follows. But he doesn't see the sd card. Writes NO SD CARD 
luc, what am I doing wrong?

------------
> [ESP780]/SD
Directory on Global FS : /SD
0 file, 0 dir
Total 0 B, Used 0 B, Available: 0 B
> [ESP780]/SD
Directory on Global FS : /SD
0 file, 0 dir
Total 0 B, Used 0 B, Available: 0 B
> [ESP790]
Incorrect command!
> [ESP790]/SD
Incorrect command!
> [ESP780]
Directory on Global FS : /
<DIR> FS
SD 0 B 2021-11-09 00:00:41
1 file, 1 dir
> [ESP200]
No SD card
> ESP700
> [ESP700]
ok
> [ESP715]FORMATSD
Start Formating
------

  config.h - ESP3D configuration file

  Copyright (c) 2014 Luc Lebosse. All rights reserved.

  This code is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This code is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with This code; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/
#ifndef _CONFIGURATION_H
#define _CONFIGURATION_H

//FEATURES - comment to disable //////////////////////////////////////////////////////////

//WiFi setup station as default, use AP mode first if not done
//Note: need both defined to enable it
//#define STATION_WIFI_SSID "*********"
//#define STATION_WIFI_PASSWORD "*********"
//you can also use a different config file for SSID/password
[... 245 lines cut ...]
//SETTINGS_IN_EEPROM 0
//SETTINGS_IN_PREFERENCES 1
#define ESP_SAVE_SETTINGS SETTINGS_IN_EEPROM

/************************************
 *
 * Customize ESP3D
 *
 * **********************************/
#if defined( ARDUINO_ARCH_ESP8266)
#define ESP_MODEL_NAME "ESP8266"
#define ESP_MODEL_URL "http://espressif.com/en/products/esp8266/"
#endif //ARDUINO_ARCH_ESP8266
#if defined( ARDUINO_ARCH_ESP32)
#define ESP_MODEL_NAME "ESP32"
#define ESP_MODEL_URL "https://www.espressif.com/en/products/hardware/esp-wroom-32/overview"
#endif //ARDUINO_ARCH_ESP32
#define ESP_MODEL_NUMBER "ESP3D 3.0"
#define ESP_MANUFACTURER_NAME "Espressif Systems"
#define ESP_MANUFACTURER_URL "http://espressif.com"

#define NOTIFICATION_ESP_ONLINE "Hi, %ESP_NAME% is now online at %ESP_IP%"
#define ESP_NOTIFICATION_TITLE "ESP3D Notification"

#endif //_CONFIGURATION_H

luc-github, 2021-11-08:
`#define SD_DEVICE ESP_SDIO //ESP_SDFAT` is incorrect you use SPI pins so you must use spi
`#define SD_DEVICE ESP_SD_NATIVE` is the proper setting

Anion-anion, 2021-11-09:
Thanks Luс!
it works!
I`am Happy)

Anion-anion, 2021-11-09:
Luс!
how to increase the speed of the sd card. Now an 8 MB file is copied for about 7 minutes - it's a long, long time. Is there a way to speed up writing and reading from the sd card by ftp?

Legsmaniac, 2021-11-09:
@Anion-anion https://blog.prusaprinters.org/easy-wireless-printing-with-flashair-sd-cards_8603/

luc-github, 2021-11-09:
@Anion-anion the esp32 webserver is not really fast at the moment
to speed upload use webdav instead and sdio for SDCard instead of SPI  and may be do some tuning : https://github.com/luc-github/ESP3D/issues/491#issuecomment-766157740

But what do you want to do with files uploaded on SDCard connected to ESP32 ? the printer cannot see it, so may I know the purpose to have fast upload on SD ?

closed 2021-11-20 by luc-github