luc-github/ESP3D #693: [BUG]<ESP32 SDIO don't support ESP_SHARED_SD >

liqijian101, 2021-11-11:
**Describe the bug**
hi luc，I test the ESP32 SDIO in ESP_SHARED_SD mode .

I add some code in  sdio_esp32.cpp ,but it did't word .   Do the   ESP32 SDIO support ESP_SHARED_SD?

uint8_t ESP_SD::getState(bool refresh)
{
    static bool lastinitok = false;
#ifdef SDMMC_FORCE_BEGIN
    lastinitok = false;
#endif //SDMMC_LIGHT_CHECK
#if defined(ESP_SD_DETECT_PIN) && ESP_SD_DETECT_PIN != -1
    //no need to go further if SD detect is not correct
    if (!((digitalRead (ESP_SD_DETECT_PIN) == ESP_SD_DETECT_VALUE) ? true : false)) {
        _state = ESP_SDCARD_NOT_PRESENT;
        return _state;
    }
#endif  //ESP_SD_DETECT_PIN
    //if busy doing something return state
    if (!((_state == ESP_SDCARD_NOT_PRESENT) || _state == ESP_SDCARD_IDLE)) {
        return _state;
    }
    if (!refresh) {
        return _state;  //to avoid refresh=true + busy to reset SD and waste time
    }
//SD is idle or not detected, let see if still the case
    _state = ESP_SDCARD_NOT_PRESENT;
    
**bool isactive = accessSD();**

//refresh content if card was removed
    if (!lastinitok) {
        log_esp3d("last init was failed try sd_mmc begin");
        SD_MMC.end();
        if (SD_MMC.begin("/SD", SD_ONE_BIT_MODE)) {
            log_esp3d("sd_mmc begin succeed");
            if (SD_MMC.cardType() != CARD_NONE ) {
                _state = ESP_SDCARD_IDLE;
                lastinitok = true;
                log_esp3d("sd_mmc card type succeed");
            } else {
                log_esp3d("sd_mmc card type failed");
            }
        } else {
            log_esp3d("sd_mmc begin failed");
        }
    } else {
        log_esp3d("last init was ok try card type");
        if(SD_MMC.cardType() != CARD_NONE) {
            log_esp3d("checking sd_mmc card type succeed");
            _state = ESP_SDCARD_IDLE;
        } else {
            lastinitok = false;
            log_esp3d("Soft sd check failed");
            //SD_MMC.end();
            if (SD_MMC.begin("/sdcard", SD_ONE_BIT_MODE)) {
                log_esp3d("new sd_mmc begin succeed");
                if ( SD_MMC.cardType() != CARD_NONE ) {
                    _state = ESP_SDCARD_IDLE;
                    lastinitok = true;
                    log_esp3d("new sd_mmc card type succeed");
                } else {
                    log_esp3d("new sd_mmc card type failed");
                }
            } else {
                log_esp3d("new sd_mmc begin failed");
            }
        }
    }
    **if (!isactive) {
        releaseSD();
    }**
    return _state;
}

bool ESP_SD::begin()
{
    log_esp3d("Begin SDIO");
    _started = true;

#ifdef SDMMC_FORCE_BEGIN
    _state = ESP_SDCARD_NOT_PRESENT;
#else
    _state = getState(true);
#endif //SDMMC_FORCE_BEGIN
**#if SD_DEVICE_CONNECTION  == ESP_SHARED_SD
#if defined(ESP_FLAG_SHARED_SD_PIN) && ESP_FLAG_SHARED_SD_PIN != -1
    pinMode (ESP_FLAG_SHARED_SD_PIN, OUTPUT);
    digitalWrite(ESP_FLAG_SHARED_SD_PIN, !ESP_FLAG_SHARED_SD_VALUE);
#endif //ESP_FLAG_SHARED_SD_PIN
#endif //SD_DEVICE_CONNECTION  == ESP_SHARED_SD**

    return _started;
}

luc-github, 2021-11-11:
the shared mode was tested on hardware that do SPI sharing, for SDIO, sharing it may follow same process, but I do not have such hardware, and so never tested it, so I did not implemented it, 
you better check the api function instead of using pinMode  :
https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/filesystem/esp_sd.cpp#L55-L72

they are used in getstate https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/filesystem/sd/sd_native_esp32.cpp#L35-L68  - so they must be added  accordingly

for SDIO the hardware switch should switch all pins involved  like for SPI but as I never tested - I cannot say

luc-github, 2021-11-11:
your changes looks correct at first sight - but what do your hardware do as switch ?

liqijian101, 2021-11-13:
Hi luc，my hardware is also use the ch440 chip.
Now,I find a way to improve the upload speed that ESP32 use SPI.
if ESP32 HSPI uses IOMUX not GPIOMUX,the spi frequence can be 80MHz,.That is higer than 26MHz.
But I don't know how to change the ESP3D code.

liqijian101, 2021-11-13:
I dont test the SDIO mode,because if esp3d uses SDIO, hower the printer boad uses SPI, it is Incompatibility.

luc-github, 2021-11-13:
May I know where do you see the SPI speed is 26MHz ?

Do you have a sample code of this GPIOMUX usage that allow better speed ? I am curious

liqijian101, 2021-11-13:
this line   :   https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html
GPIO Matrix and IO_MUX
I don't find how to use IO_MUX.  
I have the result of 26MHz , if the frequence is higher than 26MHz, the esp3d board can not find SD Card.

liqijian101, 2021-11-13:
you can search Frequency Limit  in that line.
we also can use dma to improve the speed.

luc-github, 2021-11-13:
idf allows more than arduino - or need to rebuild the core 
the SD class allow to change frequency
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/SD.h#L31

the base frequency in ESP3D is https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/filesystem/sd/sd_native_esp32.cpp#L31

closed 2021-11-20 by luc-github