luc-github/ESP3D #701: [BUG] Compiling error 3.0 platformio SD_MMC

kiss81, 2021-12-28:
For some reason the platform.io version currently included in the 3.0 branch is not working for me. I am using a fresh installed version of code 1.63.2 with the platform.io extension on Ubuntu 21.10 x64.

`In file included from /home/xxx/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.cpp:24:0:
/home/xxx/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.cpp: In member function 'bool fs::SDMMCFS::begin(const char*, bool, bool, int)':
/home/xxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/sdmmc_host.h:119:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
/home/xxx/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.cpp:45:39: note: in expansion of macro 'SDMMC_SLOT_CONFIG_DEFAULT'
     sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();`

Am I missing something?

luc-github, 2021-12-28:
according https://github.com/espressif/arduino-esp32/issues/6021
seems I missed to update platformio.ini
I just did https://github.com/luc-github/ESP3D/commit/76878d5118d1ea3d74aa169e12c4c2f8bb0f3dbf

kiss81, 2021-12-28:
Issue is solved: compiles again. Thank you! Is the 3.0 already useable for testing?

luc-github, 2021-12-28:
yes, I rewriting webui but current one is usable

kiss81, 2021-12-28:
Still running into an issue:
I am running the latest 3.0 version on an esp32 connected to an MKS robin nano with the latest Marlin firmware.
In the Marlin configuration I use:
```
#define SERIAL_PORT 1
#define MKS_WIFI_MODULE  // MKS WiFi module
```

I initially changed the COMMUNICATION_PROTOCOL to MKS_SERIAL, but that is not working for me. When I changed it back to RAW it seems to work.
I would like to have direct access to the SD card of the TFT35 or the microsd in the robin nano pcb, but I assume I need to get it working with MKS_SERIAL first?

luc-github, 2021-12-29:
Hmm it depends how your Marlin compile the FW
1 - your TFT is standalone TFT connected by serial ? or it is robin TFT whic does not have SD reader
2 - where do you connect the esp32 ? and how do you connect it ? MKS protocol use more than rx/tx but also 2 others pins

kiss81, 2021-12-29:
Thanks!
1) It's the one with a card reader and connected to a large parallel (ide style) cable to the main board.
2) I only connected the rx /tx now, so that is the reason it's not working then. Can I use the same GPIO ports for the other pins on the esp32 board as on the esp8266
I found this image:
https://www.mischianti.org/wp-content/uploads/2021/08/MKS-Robin-Nano-1.2-add-esp12-as-wifi-module-wiring.jpg

Rx and Tx are now connected to gpio16/17 on the esp32

luc-github, 2021-12-29:
> It's the one with a card reader and connected to a large parallel (ide style) cable to the main board.

So this one I do not know how you can configure SD access as in Marlin you only can access one SD Reader

about wiring yes you can use RX2/TX2 if you configure ESP3D accordingly, I see on wiring he connect SD card pins, which is weird, because they are not actually used by MKS protocol in Marlin, these pins can be used only with reprap FW

kiss81, 2021-12-29:
> So this one I do not know how you can configure SD access as in Marlin you only can access one SD Reader
>
> about wiring yes you can use RX2/TX2 if you configure ESP3D accordingly, I see on wiring he [...]

Ok great, I will figure out the wiring. But I am not sure what the MKS protocol will actually bring me. :) 
I am mainly interested in the SD card access. Maybe I am wrong, but II thought the Robin nano board supported direct (SPI) sd access? That's probably what the MISO/MOSI/CLK gpio connections are for? But I am not sure if that's supported by the es3dp firmware now? Usually the SD access is implemented over the regular Rx/Tx pins right?
Sorry for all the questions, your help is really appreciated.

luc-github, 2021-12-29:
MKS protocol do faster Serial Upload but you must not have any EMI issue or you will have noise 
As wrote, MKS protocole does not use SPI, only serial- the code is original code from DC42 to be used with RepRap but Marlin does not support this SPI implementation, so it is not used but MKS dev team left it, 
I plan to dig in to see if a porting could be done and added to esp3d also, but it is still in my huge todo list...

kiss81, 2021-12-29:
All clear to me now. I will post the final schematic amd results when one. Maybe good for the wiki?

luc-github, 2021-12-29:
sure -  wiki is actually here  https://github.com/luc-github/ESP3D/tree/2.1.x/wiki

kiss81, 2021-12-29:
I kind of got it to work now: I can list the SD card files in the PCB or the TFT. Only thing is that the upload icon is missing, so I am not able to upload new files. Any clue?

kiss81, 2021-12-29:
I compiled the (3.0 rework) WEB-UI myself and tested the marlin dist. Unfortunately the "sd printer" option is missing there so I can't view the files there.

luc-github, 2021-12-30:
you need to enable it in settings
[image]

kiss81, 2021-12-30:
It works. But still lacking the upload function. Is it depending on some configuration option in esp3d or Marlin? The 32gb sdcard is formatted fat32

kiss81, 2021-12-30:
`M115 TFT

FIRMWARE_NAME:Marlin 2.0.9.3 (Dec 29 2021 17:31:19) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Kywoo3D EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

Cap:SERIAL_XON_XOFF:0

Cap:BINARY_FILE_TRANSFER:1

Cap:EEPROM:1

Cap:VOLUMETRIC:1

Cap:AUTOREPORT_POS:0

Cap:AUTOREPORT_TEMP:1

Cap:PROGRESS:0

Cap:PRINT_JOB:1

Cap:AUTOLEVEL:1

Cap:RUNOUT:1

Cap:Z_PROBE:1

Cap:LEVELING_DATA:1

Cap:BUILD_PERCENT:0

Cap:SOFTWARE_POWER:0

Cap:TOGGLE_LIGHTS:0

Cap:CASE_LIGHT_BRIGHTNESS:0

Cap:EMERGENCY_PARSER:0

Cap:HOST_ACTION_COMMANDS:0

Cap:PROMPT_SUPPORT:0

Cap:SDCARD:1

Cap:REPEAT:0

Cap:SD_WRITE:1

Cap:AUTOREPORT_SD_STATUS:0

Cap:LONG_FILENAME:0

Cap:EXTENDED_M20:0

Cap:THERMAL_PROTECTION:1

Cap:MOTION_MODES:0

Cap:ARCS:1

Cap:BABYSTEPPING:1

Cap:CHAMBER_TEMPERATURE:0

Cap:COOLER_TEMPERATURE:0

Cap:MEATPACK:0

Cap:CONFIG_EXPORT:0

ok`

luc-github, 2021-12-30:
it is normal https://github.com/luc-github/ESP3D/discussions/540#discussioncomment-1604767

luc-github, 2022-01-06:
i guess issue can be closed now

closed 2022-01-06 by luc-github