luc-github/ESP3D #702: [Question] Chitu v6 board and Chitu Wifi module

KingBain, 2021-12-28:
On the stock chitu v6 board(common board in TronXY X5sa pro printers) the pinout is this:

[image]

 The pinout 
 
[image]

From factory, Pin 5 is pulled down to ground. According to the documentation here, 

Pin 5 is GPIO0 on the Tronxy Wifi modules, which is basically an ESP01 with 1MB of from.

so if Pin 5 in pulled down to ground, this means the board will never boot up because ESP8266 is being put into upload mode. 

[image]

Am I understanding that correctly ?

If I soldered a 10K resistor between pin 5 and pin 8, would that pull pin 5 up ?

Also from factory Pin 4 (CH_PD) is getting 5v .... how bad is that for the ship ?

luc-github, 2021-12-29:
are you sure they use a ESP8266-01 connector ? put GPIO to GND is not normal at start
also I see it have 5V output which is not supported by any ESP chips, ESP8266 have a tolerance but it is not recommended at all

it looks more a custom connector - for custom esp8266-01 board

Thomaskru, 2021-12-29:
Hi!
I have also this board (tronxy x5sa-2e). ESP8266 is not 1to1 compatible with this pinout.

You can use male/female Dupont wires to lift the esp of this socket. Connect pins
(Chitu -> ESP8266)
1 GND -> 1 GND
2 WIFI_TX -> 2 TX
7 WIFI_RX -> 7 RX
8 3V3 -> 8 3V3
 via wires.

Additionaly you must shorten 3V3 & EN (Pin 8 & 4) on the esp8266 to force boot. 

You can then use pins gpio0 & gpio2 from the Chitu board for other things like  bltouch or something else. (Marlin only).

ESP32 is also possible with these RX/TX Pins and when it has support for 5V you can use pin 4 from chitu board to power ESP32.

KingBain, 2021-12-29:
I ended up just bending the GPIO pin out of the way and inserting the wifi module into the socket on to the board. 

The ESP3d then successfully booted.

I also found this on a Marlin forum talking about the v6 pinouts. I had to translate it from Chinese to english 
[image]

it looks like they're aware of the v5 issue... I guess they just run the chip hot :)