luc-github/ESP3D #714: [Question]SD Files empty is it, is it different from SD card in printer?

BrandonOh, 2022-01-30:
So I am using a Kywoo 3d printer and got the ESP3D set up but when ever I try to upload a file to the printer from the interface it uploads very slowly and then about half way I get an error saying no connection. I decided that since the upload is slow ill try putting an SD card into the printer and see if I can at least tell it to print from the interface, but the SD Files screen says that there are no files. Is the SD Files a separate card? or is my interface bugging out? I have two separate SSD and SD cards in my printer but nothing is still showing up. Any help would be appreciated or an alternative to uploading a file to my printer as my handicap makes it hard to move back and forth to my printer.

luc-github, 2022-01-30:
First yes serial upload is slow like hell and subject to lot of issues if anything disturb the upload it is pinned issue https://github.com/luc-github/ESP3D/discussions/535

you mention you have 2 SD cards readers : 
1 - where are they located ?
2 - what is the output when you type  `M20`  in webui terminal ?

BrandonOh, 2022-01-30:
> First yes serial upload is slow like hell and subject to lot of issues if anything disturb the upload, it is pinned issue #535
> 
> you mention you have 2 SD cards readers : 1 - where are they located ? 2 - what is the output when you type `M20` in webui terminal ? 3 - what is the output when you type `M115` in webui terminal ?

Thank you for the response.

1. I currently have two SD cards inserted into my printer. I have one micro SD card and one micro SD to SD adapter card inserted. I was unsure if the ESP3D can see those SD cards connected into the printer or if there is a whole separate spot for it to be insterted.
2. The M20 output was:
```
M20
Begin file list
End file list
ok
```
3. The M115 output was:
```
M115
p:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
FIRMWARE_NAME:Marlin 2.0.9 (Aug  9 2021 07:40:19) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Kywoo3D EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:HOST_ACTION_COMMANDS:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
ok
```
hope that answered the questions you had for me.

luc-github, 2022-01-30:
seems your printer does not see the content of SD card neither as M20 give empty sd 

>I currently have two SD cards inserted into my printer. I have one micro SD card and one micro SD to SD adapter card inserted.

Sorry this does not answer - where are these SD reader located - do you have picture ? 

Only Smoothieware support 2 sd card reader as far I know - and you are using Marlin - so only one SD is visible from Marlin and this one look empty of GCODE files 

Check which one has .GCO file on it and do M20 to see if they are seens

BrandonOh, 2022-01-30:
> seems your printer does not see the content of SD card neither as M20 give empty sd
> 
> > I currently have two SD cards inserted into my printer. I have one micro SD card and one micro SD to SD adapter card inserted.
> 
> Sorry this does not answer - where are these SD reader located - do you have picture ?
> 
> Only Smoothieware support 2 sd card reader as far I know - and you are using Marlin - so only one SD is visible from Marlin and this one look empty of GCODE files
> 
> Check which one has .GCO files on it and do M20 to see if they are seens

So after reading your reply I took out one of two SD cards and I think that was the problem because now I can see the list and the M20 command also showed a list. Thank you for the help!

closed 2022-01-30 by BrandonOh