luc-github/ESP3D #715: [FEATURE REQUEST]Setup Mode

luc-github, 2022-02-02:
As now more people want to use AP mode instead of Client Mode there are some issues raised, because current AP mode is actually made for setup and not for full usage, it will be quickly overloaded, indeed current AP mode is made to catch every query / requests and handle them which lead to trouble if intensive bandwith is used.

So I will add a new mode : Setup mode, and modify the  current AP mode, here the feature list of each mode for reference

Mode | Setup (Access point) | AP  (Access point)| STA (Client)
------|-------------------------|--------------------|-----------
Usage| Setup only | Normal | Normal
Captive Portal | Yes | No | No
mDNS | No | No | Yes
SSDP | No | No | Yes
Gateway | Yes (Self) | No |  Yes  (AP)
DNS | Yes (self) | No | Yes (AP)
 
If Client mode failed it should go to Setup mode.
 
 Feedback are welcome

luc-github, 2022-02-06:
Here the workflow of wifi mode according boot process 
[image]

closed 2022-02-06 by luc-github

ETE-Design, 2022-02-06:
Nice done, will it also be possible to change between the modes with "Direct ESP Commands"

luc-github, 2022-02-06:
Yes it is implemented - but I did not updated wiki yet
```
[ESP104](State) - display/set sta fallback mode which can be BT, SETUP, OFF
[ESP110](State) - display/set radio state which can be BT, WIFI-STA, WIFI-AP, WIFI-SETUP, ETH-STA, OFF
```

ETE-Design, 2022-02-08:
@luc-github Is [ESP104] SETUP and [ESP110] WIFI-SETUP almost the same? Only thing that [ESP104] SETUP have is a Fallback if it can't connect?

commit a4609aee (2022-02-06): Add new Setup Mode and new AP mode, fixes #715