luc-github/ESP3D #722: [BUG]<Enter comprehensive title>

X3msnake, 2022-02-22:
**Describe the bug**
exit status 1
Error compiling for board AI Thinker ESP32-CAM.

ResolveLibrary(hwcrypto/sha.h)
  -> candidates: []C:\Users\Utilizador\Documents\Arduino\libraries\arduinoWebSockets\src\WebSockets.cpp:42:10: fatal error: hwcrypto/sha.h: No such file or directory
 #include <hwcrypto/sha.h>

**To Reproduce**
Steps to reproduce the behavior:

Followed the instruction on the page, copied the libraries and at uploading it gives me the above error

AI-Thinker ESP32-CAM, Arduino IDE 1.8.19 - Windows 10

luc-github, 2022-02-22:
what version of the esp32 core are you using ?

luc-github, 2022-02-22:
2.0.2 is correct it use idf 4.4

luc-github, 2022-02-22:
so I am guessing you do not use the proper websocket arduino library version - you use the one present in repository of 2.1 whhich is 2.1.0 for core 1.X 
the one I have put in 3.0 is core core 4.X it is version 2.3.5 https://github.com/luc-github/ESP3D/tree/3.0/libraries/arduinoWebSockets-2.3.5
https://github.com/luc-github/ESP3D/blob/3.0/libraries/arduinoWebSockets-2.3.5/src/WebSockets.cpp#L44-L48

you must match librairies versions present in each branch

X3msnake, 2022-02-22:
Ok, i'll try compiling 3.0 with the libraries you mentioned installed now

X3msnake, 2022-02-22:
It worked, thanks. sent you a token of my appreciation for the project via paypal. Best regards

[image]

closed 2022-02-22 by X3msnake

luc-github, 2022-02-23:
@X3msnake thank you ^_^