luc-github/ESP3D #723: [Question]<ESP32-CAM SD card, cannot see any files>

X3msnake, 2022-02-25:
Greetings Masters

Finally managed to install version 2.1.1 after battelling with the libraries, ended up uninstalling and resetting the arduino IDE and starting all from scratch folling the instrutions.

Now the UI is working and it is connecting to the board. however i am using ESP32-CAM and was hoping i could use the SD card it has.

Is there any special config one must do to use the card on this hardware?

luc-github, 2022-02-26:
esp32 camera is supported in ESP3D 3.0
Direct SD card connection is supported in 3.0 but streaming GCode is not yet implemented so SD usage is currently limited to configuration file storage.

X3msnake, 2022-02-26:
Oh :|

In that case I cannot use it untill then. Was hoping i could use it as a GRBL streamer since the board on my 3018 Vevor CNC machine farm do not have any native SD card support like on the 3DPrinters

Any ETA on the implementation of streaming from SD card for the ESP32-CAM?

luc-github, 2022-02-26:
No sorry but you can check update here : https://github.com/luc-github/ESP3D/issues/243

closed 2022-02-28 by luc-github