luc-github/ESP3D #729: [BUG]"Error (12) Upload Rejected" While Trying To Upload to SD Card

FergyA, 2022-03-10:
I am running ESP3D on an ESP8266 connected to a BTT TFT35 E3 V3.0, running on a BTT SKR Mini E3 V3 running Marlin 2.0.x bugfix. The firmware of all 3 devices have been freshly compiled from source today 3/9/2022.

Whenever I try to upload gcode through the webui to the SD card I get the following error. 
[image]
In addition, the printer throws a random number (usually 1-3) of "Line number is not Last Line Number +1" errors. The file is appropriately named in 8.3 format, and size does not matter.

All other features of the webui work, including reading/starting a print from the SD card, controlling temperatures, and issuing commands. I am even able to create a file and populate it manually through the command terminal.

I did also try enabling the DISABLE_SERIAL_CHECKSUM option in the firmware, which seemed to eliminate the line number errors, but the same upload still fails.

```
[ESP800]
FW version:2.1.1.b8 # FW target:marlin # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:no # webcommunication:Sync:81:192.168.1.83 # hostname:ESP3D
```

```
[ESP420]
{"chip_id":"3141289","cpu":"160","freemem":"36.96 KB","SDK":"2.2.1(cfd48f3)","flash_size":"1.00 MB","update_size":"376.00 KB","spiffs_size":"228.20 KB","baud_rate":"115200","sleep_mode":"Modem","channel":"7","phy_mode":"11g","web_port":"80","data_port":"8888","hostname":"ESP3D","active_mode":"STA (40:F5:20:2F:EE:A9)","connected_ssid":"Ferguson","connected_signal":"100%","ip_mode":"DHCP","ip":"192.168.1.83","gw":"192.168.1.1","msk":"255.255.255.0","dns":"192.168.1.1","disabled_mode":"AP (42:F5:20:2F:EE:A9)","captive_portal":"Enabled","ssdp":"Enabled","netbios":"Disabled","mdns":"Enabled","web_update":"Enabled","pin recovery":"Disabled","autentication":"Disabled","target_fw":"Marlin","M117_output":"Enabled","Serial_output":"Enabled","Websocket_output":"Enabled","TCP_output":"Enabled","fw":"2.1.1.b8 ESP8266/8586"}
```
```
M115
SD_WRITE:1
FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar  9 2022 20:51:56) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Pro EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:1
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:1
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:1
Cap:PROMPT_SUPPORT:1
Cap:SDCARD:1
Cap:MULTI_VOLUME:0
Cap:AUTOREPORT_SD_STATUS:1
Cap:LONG_FILENAME:1
Cap:LFN_WRITE:0
Cap:CUSTOM_FIRMWARE_UPLOAD:0
Cap:EXTENDED_M20:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
Cap:CONFIG_EXPORT:0
area:{full:{min:{x:0.0000,y:0.0000,z:0.0000},max:{x:235.0000,y:235.0000,z:250.0000}},work:{min:{x:0.0000,y:0.0000,z:0.0000},max:{x:235.0000,y:235.0000,z:250.0000}}}
ok
```

FergyA, 2022-03-10:
Also I have tried several different SD cards, and made sure they are properly formatted with FAT32.

luc-github, 2022-03-10:
Yes I see several things that are unfortunatly not compatible with M28/M29 upload and so make upload more likely failing 
the upload is failing since the begining 12 = ESP_ERROR_START_UPLOAD

1 - you have auto report enabled 
When upload is ongoing Marlin continue to send report which disturb the upload, I can see it with the error `Line number is not Last Line Number +1` 
2 -  You have a TFT which also send queries - it seems since a while the PR I pushed that disable polling when upload is ongoing is broken and so in same time upload the TFT also send commands which will disturb also upload

You can try to confirm my guess by reflashing Marlin disabling all auto reports but for BTT TFT I am not sure how you can workaround as ESP board is connected to it.
 
Also because this protocol is not really used, it is often broken in Marlin / TFT because people do not test it or even consider it when adding auto report features, polling commands.

So there is not much I can do currently in ESP3D to prevent others commands to disturb upload and make it failed

I am working on another protocol because current M28/M29 transfer protocol only bring disappointment in performance (0.4B/sec) https://github.com/luc-github/ESP3D/discussions/535  and reliability (cannot handle concurrent commands as you can experience it  / Marlin do not handle the stop auto report when upload is ongoing) : https://github.com/luc-github/ESP3D/issues/575

FergyA, 2022-03-10:
Thanks for the quick and detailed response!

I did try recompiling without auto reports enabled and unfortunately it didn't seem to resolve anything. Still get error 12, and still get Line number is not X errors on the TFT, so I suspect your guess that the TFT is interrupting transfers is correct.

In the end given the transfer speeds I guess I'll just give up and get a wireless SD card and just use ESP3d for monitoring prints since everything else works fine.

luc-github, 2022-03-10:
@FergyA to be honest I did support the M28/M29 at the beginning just for courtesy, I also use WiFi SD and WifiFIback pack for the SD transfer.
FYR in new version of ESP3D this old protocol will not be supported anymore

FergyA, 2022-03-10:
I just realized... this youtube video demonstrates almost my exact setup, and yet he is able to upload. Any comments on what makes his setup work even though he's routing through the TFT?

https://www.youtube.com/watch?v=AGTF-JEUHNA

luc-github, 2022-03-10:
yes it was one year ago, when I do the pr to btt FW it was 2 years ago and was also  working

as I mentioned Marlin and BTT fw break the protocol doing polling / report status when M28/M29 is ongoing

check what doing these polling/status and make them quiet when M28/M29 isprocessing is the way to go

FergyA, 2022-03-10:
Got it, thank you! I'm kind of tempted to figure out what those firmware versions were and see if I can rebuild them then. There are no new major additions in the last year to either the TFT, Marlin, or ESP3D that I think I would miss,

Appreciate the help!

closed 2022-03-10 by FergyA