luc-github/ESP3D #730: [Question] settings for flashing ESP32 version with flash download tools 3.6.7

digant73, 2022-03-16:
I compiled ESP3D for my ESP32 dev board using `env:esp32dev` as the target in platformio.ini. I'm using "flash download tools 3.6.7" to flash the created files `firmware.bin` and `partitions.bin` but I need the settings to use (e.g. base address for files, SPI Mode (is it QIO?), Flash size (my board is 32Mbit) etc...). Is it possible to provide them?

luc-github, 2022-03-16:
I think it would be easier to get the upload info/missing binaries by using arduino ide and esp32 tools directory of esp32 package

It is great you did such PR, ignoring check sum should not be a problem if serial conection is ok. 
Did you check the reliability and speed ?

digant73, 2022-03-16:
yes, the reliability was ok. I set 500000 as serial speed. BTT fw allows also a speed of 1000000 (as in Marlin fw) but ESP3D allows a maximum of 921600. If 1000000 could be configurable also in ESP3D it should be better (I would try and use that speed)

luc-github, 2022-03-17:
Yes because these speed are not well supported : https://github.com/luc-github/ESP3D/issues/350#issuecomment-511133797
you can also see that even it looks correct, it may not if you do not proper comparaison, - serial can corrupt data if too fast
https://github.com/luc-github/ESP3D/issues/296#issuecomment-460858342 which may not pop up to eyes as GCODE is text file
so final resul may be hazardous

digant73, 2022-03-17:
understood, thanks!