luc-github/ESP3D #733: [BUG]Use ESP3D firmware on MKS Robin WIFI

CLP58, 2022-04-12:
I want to use ESP3D on the MKS Robin nano V3
I have added the following to ESP 3.0.
Config the MKS uart mode on file of "Configuration.h"
#define COMMUNICATION_PROTOCOL MKS_SERIAL
I use VS code
An error has occurre
[image]
d

`        } else {
            //not supported
            log_esp3d("Mode not supported : %d ", NetConfig::getMode());
            return false;
        }
        } else {
            //not supported
            log_esp3d("Mode not supported : %d ", NetConfig::getMode());
            return false;
        }
else if (NetConfig::getMode() == ESP_WIFI_AP || (NetConfig::getMode() == ESP_AP_SETUP)`

And the "myconfig.h" file cannot be found
I'm a newbie, can anyone tell me what I'm doing wrong?

CLP58, 2022-04-12:
The problem has been solved, due to the spelling in mks_service.cpp. Thanks to lus！
[mks_service.zip](https://github.com/luc-github/ESP3D/files/8469742/mks_service.zip)

closed 2022-04-12 by CLP58

luc-github, 2022-04-12:
Fixed by https://github.com/luc-github/ESP3D/commit/b3f2e46b8646648a297d5cbf0b32e34ceb8b5f40 thank you