luc-github/ESP3D #737: [BUG] Readme needs to be refactored

atsju, 2022-04-18:
As a first time user i see things with a different eye than experiences users.
IMO the readme is unclear. For installation it should rely completely on wiki (today there is information in both).
I can't refactor it myself without discussing with some user that knows everything.

luc-github, 2022-04-18:
I fully agree - but most of people do not read the wiki and even not check the readme - they prefer to watch youtube video which is almost all the time outdated, when people say they cannot do even reading the readme/wiki and I ask what did they did not understood - so I can improve it - they usually never answer ...

spend time on a full documentation that people do not take time to read vs spend time on code and fix bug => I did my choice

This also why I tried to move to platformio - it simplify a lot the installation / compilation steps - all is ready, just update the configuration.h, which I try to fill with lot of comments, and choose the target board ....this is the purpose of https://github.com/luc-github/ESP3D/issues/478

Using arduino is more tricky for the installation. but arduino is also more up to date for board / chip support

That said every contribution for documentation is welcome, and I would be more than happy to get nice documentation - but it won't come from me until my todo list is done ( in some years I guess 😸 ), it is definitly not my priority

atsju, 2022-04-18:
I understand the fact of moving to platformIO. Marlin users already are on platformIO so I think it's good to have same tools.

_If_ I was to rewrite the documentation, is the platformIO version ready and recommended with 2.1.1 or shall 2.1.1 always build with Arduino ?

luc-github, 2022-04-18:
yes it is : https://github.com/luc-github/ESP3D/blob/2.1.x/platformio.ini

atsju, 2022-05-08:
@luc-github I'm still in the learning curve. What are data port and web port for ? I understand he web port, my question is more, is the data port actually used and when ?

luc-github, 2022-05-08:
some application like pronterface or repetier host allow direct tcp communication, this is the port fort this usage, several people call this telnet  https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WiFiTelnetToSerial but is actually not right, it is more raw tcp, so I have called data port in 2.x, but in 3.x I is called telnet port because it is more meaningful for people

atsju, 2022-05-09:
note for update of wiki
Do a section about voltage adaptation between MCUs.
This picture might be updated to put only a serie resistor to limit current. Better and easier than a divider bridge.
[image]

luc-github, 2022-05-09:
I did not do the picture - my electronic skills are very limited - so I have just published what people did and experiment by themself

luc-github, 2022-05-27:
@atsju Is this issue still valid as all PR have been processed already ?

atsju, 2022-05-27:
For me yes still valid. I plan to help, I had no computer acces for 2 weeks. I want to refactor readme and wiki gradually.

atsju, 2022-05-27:
@luc-github  https://github.com/luc-github/ESP3D/blob/2.1.x/wiki/last_update.txt seems never to update. Remove file or fix something automated ?

luc-github, 2022-05-27:
was to trigger the copy but this seems no more necessary

atsju, 2022-05-29:
@luc-github I want to build for ESP32cam (and later I want to update the wiki for futur 3.0 version also)

I built 3.0 branch (not refactor) with platformIO. I built with `[env:esp32cam]` env.
I see no wifi network and nothing relevant on COM port.
Also my board seems to be based on ESP32-S2 and for this setting I read 
```
[env:esp32-S2]
;EXPERIMENTAL AND NOT WORKING FOR ME
```

any advise ?(I have found some site stating that ESP32-CAM is a module from AI thinker and not espressif. But my ESP32CAM is based on an espressif module.Could this be an issue ?

luc-github, 2022-05-29:
I do not think there is any ESP32Cam based on S2 (which is mono core), they are using ESP32 or ESP32S which is same chip 
Some ESP32Cam need to be powered using 5V to work, the 3.3v is just for flashing
AT thinker is common brand name and it is most of basic ESP32Cam as far I know

atsju, 2022-05-29:
Module is powered though USB 5V with onboard regulator on the daughter board. Should be fine.

atsju, 2022-05-29:
Thanks. Had something to do with reset I suppose.
Sorry I couldn't copy paste text...
I seems the serial communication fails a bit at startup (in boot I suppose). Works better after application starts
[image]

luc-github, 2022-05-29:
that is normal boot check this to suppress boot loader
https://github.com/luc-github/ESP3D/discussions/182

 like I said everything is in FAQ that no one read ^_^

atsju, 2022-05-29:
Once again thank you
and https://www.youtube.com/watch?v=LE0aehh52xU
Even having used several hours here, I still feel so lost about this repo.

about the 3.0 version (I didn't try 3.0refactor)

**I have no more setup wizard like I had in 2.x. Is it normal ?** uploaded the index.html.gz from ESP32 repo (from 3.0).

also some additional things that may need separate tickets ?:

- is it possible to add test "please upload index.html.gz" when there is no such file ? (was the case is 2.x version)
- is it possible to fail explicitly when password is wrong admin/admin ? (my phone puts uppercasse A. even if I knew, I wanted to see what happened) like "wrong username or password"
- is it possible that the field doesn't make android kayboard think it's a word/sentense starting with uppercase ?
- is it possible to make "enter" of phone keypad click on "login" button ?

luc-github, 2022-05-29:
> have no more setup wizard like I had in 2.x. Is it normal ? 
* the wizard is not yet done in 3.0 

>is it possible to add test "please upload index.html.gz" when there is no such file ? (was the case is 2.x version)
* I did in 2.1 and got feedback => I do not understand - so I removed it as in readme the process will be explained 

>is it possible to fail explicitly when password is wrong admin/admin
* I forget what is displaying - but error is 404 without details if username or password

>is it possible that the field doesn't make android kayboard think it's a word/sentense starting with uppercase
* not sure what you means but looks like autocorrection settings from your phone

>is it possible to make "enter" of phone keypad click on "login" button ?
The current field API for modal dialogs  does not have such feature - I need to see how to add it

atsju, 2022-05-29:
> > have no more setup wizard like I had in 2.x. Is it normal ?
> 
> * the wizard is not yet done in 3.0

OK so I will have to learn manual commands. Thanks

> > is it possible to add test "please upload index.html.gz" when there is no such file ? (was the case is 2.x version)
> 
> * I did in 2.1 and got feedback => I do not understand - so I removed it as in readme the process will be explained
> 

I'm sure the feedback is good. I did not understand at the beginning also. But I'm unsure of the chosen solution "removing it completely". Well It's an idea, you have more feedback than I do so you are the boss here.

> > is it possible to fail explicitly when password is wrong admin/admin
> 
> * I forget what is displaying - but error is 404 without details if username or password

From a user point of view the 404 or anything is not relevant. What happens is that the page reloads (I suppose) and it goes fast enough not to understand why. I recommend an explicit message (if possible as I don't know the limitations of all the system)

> > is it possible that the field doesn't make android kayboard think it's a word/sentense starting with uppercase
> 
> * not sure what you means but looks like autocorrection settings from your phone

yes has to do with autocorrection. I do not know HTML at all (I do only embedded) but I though something like 
[this](https://stackoverflow.com/questions/23853473/how-to-disable-autocorrect-in-html-input-on-android)

> > is it possible to make "enter" of phone keypad click on "login" button ?
> 
> * The current field API for modal dialogs  does not have such feature - I need to see how to add it

thanks

luc-github, 2022-05-29:
about autocorrect /autocomplete I did not know these settings, I can add them as actually I already set
`spellcheck="false"`

luc-github, 2022-05-29:
The authentication seems broken in rewrite I will have a look tomorrow - I need to fix it first , but before I have another bug to fix orz

atsju, 2022-05-29:
> The authentication seems broken in rewrite I will have a look tomorrow - I need to fix it first , but before I have another bug to fix orz

OK I will use without then. Thanks for feedback

atsju, 2022-05-29:
how do I setup 3.0 to go to STA ?
currently AP is working, uploading index.html.gz seems to have no effect but well it's there.
I tried setting ESP100 + ESP101 with my box settings and then `[ESP110]WIFI-STA` 
[image]
i'm on b3f2e46b8646648a297d5cbf0b32e34ceb8b5f40

luc-github, 2022-05-29:
as you update from 2.1 to 3.0 it is better to do a [ESP444]RESET to clear the EEPROM

atsju, 2022-05-29:
my esp32-CAM only has seen 3.0. Anyway I tied RESET and it did not change anything.

luc-github, 2022-05-29:
why using static IP ?

luc-github, 2022-05-30:
I am on authentication now and I understand your suggestions:
There are 2 pages I need to work 
1 - embedded page
2 - webui

# about embedded page:
[image]
The code is small on purpose but I think add message of invalid credential and enter on password should be easy and not code consumming
as I only get 404 error I will check if user already enter user id  to detect if first login or invalid credential
I will try to set like this : enter on user id will focus password, enter on password will same a click login

# about webui 
[image]

it seems the test server does not handle authentication properly as it is ok on real FW, but I will try same process as above  

I will work on these today

atsju, 2022-05-30:
> why using static IP ? why putting same ip as you box ?

No I use DHCP. and my box is on 192.168.1.0 I just tried this because I was searching a solution and was wondering if default 3rd number could prevent it from working. This is not working.

Please note, after uploading inder.html.gz, I do not get the webUI but still the embedded page. Might be related to my issue ?

thanks for the modification proposal on authentication. I will try them when you are finished.

also should I use 3.0refactor or 3.0 ? (of course I already tried both)

luc-github, 2022-05-30:
do a screen shot of the screen after upload as I do not understand your issue
The authentication modification are done in https://github.com/luc-github/ESP3D/tree/3.0-refactoring
and https://github.com/luc-github/ESP3D-WEBUI/tree/3.0-rewrite
I was suposed to merge both to respective branch 3.0 today but I found the Screen Oled support is broken - it need a library update on esp32 and actually a big code cleaning - so I am on it now 
I hope all to be ok for wednesday as I will reorganize the display code which is messy right now

luc-github, 2022-05-30:
[image]
here a screencast I did today to show how to setup esp32cam  FYR

atsju, 2022-05-30:
I will check later when coming home. thanks for screencast.

atsju, 2022-05-30:
Good news : I suppose I didn't have the right index.html.gz before
I tried https://github.com/luc-github/ESP3D/tree/3.0-refactoring and https://github.com/luc-github/ESP3D-WEBUI/tree/3.0-rewrite with the index.html.gz from marlin directory and this seems to work. 

Questions from what I discovered : why are there so many different index ? what is difference between marlin / marlin-embedded and so on ? also this file in particular is 15month old https://github.com/luc-github/ESP3D-WEBUI/blob/3.0-rewrite/dist/index.html.gz . Is it normal that it is still there (I took it by error and of course it didn't work)

And my main problem beeing impossible to connect to my box as a client is still same :(

luc-github, 2022-05-30:
Marlin is using ESP3D, Marlin-embedded is using ESP3DLib for ESP32 as main board, the commands are the same in this case minus several features like MKS protocol and some others when ESP3D & ESPSPDLib are at same level of code.
Marlin / Repetier / Smoothieware/Grbl are different firmwares, using different GCODE, and different format of response, WebUI 2.1 was a mess of `if this FW then else if this FW then..etc` this way simplify code, allow any customization for one FW without affecting others,, dicrease the final size of package  removing the unused code because at the end is only for one FW.
To sum up it simplify a lot 

The 2.1.1 is 108 KB when 3.0 refactoring is 79KB benefit is visible

luc-github, 2022-05-30:
>And my main problem beeing impossible to connect to my box as a client is still same :(
the only issue I can see is :
the SSID has some characters the ESP do not like, I do not limit any so issue is in core
the password has some characters that ESP3D doe not like for some reason - do you have special chars in your password ? others than letter and number ?

atsju, 2022-05-31:
Thanks for all these explanations. 

SSID has a _ chat but it worked on 2.1 and password is completely normal chars with letters and numbers. I will try the sample code when I get time to do so.

atsju, 2022-06-06:
I don't know what I did wrong but I made everything clean again and this time it worked. I suppose writing the wiki helped me understand what to clean and what index to upload.

Also, 3.0 are only alpha so I will not complain, but please heads up about how you want to manage index.html.gz. I see several solutions but today the alphas have only a line telling to go to ESP3D-WEBUI. So in the futur you might want one of these:
- Use github submodule so the ESP3D repo always points to correct ESP3D-WEBUI (not easy for new users)
- copy the compatible index to ESP3D github release assets (this is easy for everybody)
- copy the compatible index to ESP3D repo before release (would need new ESP3D release on update of index)

luc-github, 2022-06-06:
Now the webUI is different according target  it will be more difficult 
I guess the 2nd would be a good choic but there won't have one index.html  but the full set, the wizard should help to update if meet my expectations

atsju, 2022-06-06:
you could put everything in a zip and add it to assets (to have folder to differentiate all files with same names)

luc-github, 2022-06-06:
>you could put everything in a zip and add it to assets (to have folder to differentiate all files with same names)

this is case already : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist

atsju, 2022-06-06:
Yes but this is additional steps, user must understand the need to go to WEBUI repo (for new user this is NOT immediate) and the search into folders (dist is a good folder name but you still need to search a bit).
Adding a direct link to release folder is OK I suppose. https://github.com/luc-github/ESP3D-WEBUI/tree/V3.0-alpha2/dist maybe you can edit the release text.

btw ESP32cam is unable to connect to network as client. I tried the link with demo you shared. I'm investigating and will let you know when I need more help

atsju, 2022-06-06:
@luc-github sorry to bother you again...

So I did several test on my ESP32CAM board. It's [this](https://fr.aliexpress.com/item/32981449079.html) one and picture is accurate with what I got. Notice the logo is not AI thinker even if I think it's a 1:1 copy of the AI thinker model

I'm on realease 3.0-alpha2. The release works fine on my ESP8266 board.
As already talked about, the code on ESP32CAM works as AP only. but when trying to be client mode it fails to connect.
I tried you recommandation and the sample code doesn't connect either.

> If wifi dhcp sample work : https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientBasic/WiFiClientBasic.ino

And this doesn't work either.

I tired several things, erasing flash, changing platformIO.ini to `board = esp32cam` ... nothing seems to have an effect.

Additional infos from ESP-tool : 

```
C:\Users\xxx\Downloads\esptool-v4.1-win64>esptool.exe  chip_id
esptool.py v4.1
Found 7 serial ports
Serial port COM9
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:c2:a0:40
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 0c:b8:15:c2:a0:40
Hard resetting via RTS pin...

C:\Users\xxx\Downloads\esptool-v4.1-win64>esptool.exe  flash_id
esptool.py v4.1
Found 7 serial ports
Serial port COM9
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:c2:a0:40
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

Do you have any idea ?

luc-github, 2022-06-06:
this camera has connector for antenna - the connection to the embedded antena may be not ok
did you check the signal strengh of your router seen by the esp board ?
`[ESP420]`

atsju, 2022-06-06:
There is resistor for choosing between connector and F antenna. 
The F antenna (PCB antenna) is used in my case. Also the client mode works fine and the board is just near the box so except a short circuit like I had on my crappy board, it should connect (even with no antenna connected).

Anyway to double check wifi analyser gives good signal (when ESP is AP) about -53dBm (my box is -32dBm for reference). Not excellent but not bad.

As ESP doesn't connect to router, here is just the output of [ESP420] when phone is connected to AP

```
;chip id: 16544
;CPU Freq: 240Mhz
;CPU Temp: 41.7C
;free mem: 190.71 KB - PSRAM:4.00 MB
;SDK: v4.4.1-1-gb8050b365e
;flash size: 4.00 MB
;size for update: 1.87 MB
;FS type: LittleFS
;FS usage: 88.00 KB/192.00 KB
;baud: 115200
;sleep mode: none
;wifi: ON
;hostname: esp3d
;HTTP port: 80
;Telnet port: 23
;ap: ON
;mac: 0C:B8:15:C2:A0:41
;SSID: ESP3D
;visible: yes
;authentication: WPA2
;DHCP Server: ON
;ip: 192.168.0.1
;gw: 192.168.0.1
;msk: 255.255.255.0
;clients: 1
;# 0: 192.168.0.2(20:F4:78:FD:FC:2E)
;sta: OFF
;mac: 0C:B8:15:C2:A0:40
;serial: ON
;notification: ON(none)
;Target Fw: unknown
;FW ver: 3.0.0.a200
;FW arch: ESP32 (PSRAM)

```

atsju, 2022-06-06:
If it helps, I built example with CORE_DEBUG_LEVEL=5

```
[  3641][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 18
[  3641][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  3646][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  3650][I][WiFiMulti.cpp:119] run(): [WIFI] 18 networks found
[  3656][D][WiFiMulti.cpp:149] run():  --->   0: [11][12:34:56:78:9A:BC] SFR_AB12 (-49) *
- privacy removed the other APs -
[  3802][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 12:34:56:78:9A:BC SSID: SFR_AB12 Channel: 11 (-49)
[  3814][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  6661][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: SFR_AB12, BSSID: 00:00:00:00:00:00, Reason: 201
[  6661][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  6669][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  6675][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
[  6682][E][WiFiMulti.cpp:181] run(): [WIFI] Connecting Failed AP not found.
```

I don't see anything relevant. Will google more.

luc-github, 2022-06-06:
for your reference mine :
```
--- Miniterm on COM18  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
;Setup wifi
;Connecting
;192.168.2.205
;192.168.2.205
[ESP420]
;chip id: 50221
;CPU Freq: 240Mhz
;CPU Temp: 42.8C
;free mem: 162.10 KB - PSRAM:3.91 MB
;SDK: v4.4.1-1-gb8050b365e
;flash size: 4.00 MB
;size for update: 1.87 MB
;FS type: LittleFS
;FS usage: 92.00 KB/192.00 KB
;baud: 115200
;sleep mode: none
;wifi: ON
;hostname: esp3d
;HTTP port: 80
;Telnet port: 23
;camera name: ESP32-CAM(4)
;sta: ON
;mac: 24:0A:C4:BB:2D:C4
;SSID: luc-ext1
;signal: 50%
;phy mode: 11n
;channel: 3
;ip mode: dhcp
;ip: 192.168.2.205
;gw: 192.168.2.1
;msk: 255.255.255.0
;DNS: 192.168.2.1
;ap: OFF
;mac: 24:0A:C4:BB:2D:C5
;serial: ON
;notification: ON(none)
;Target Fw: unknown
;FW ver: 3.0.0.a200
;FW arch: ESP32 (PSRAM)
[ESP410]
;Start Scan
;luc-ext1       44%     Secure
;End Scan
```

atsju, 2022-06-06:
> NO_AP_FOUND - I still guess signal is too low - it cannot contact the AP My chip is same as you - it is AI-Thinker ESP32-S What [ESP410] give you ? it is scan with signal strengh

The example scan with log gives -49dBm it's reasonable
`[  3656][D][WiFiMulti.cpp:149] run():  --->   0: [11][12:34:56:78:9A:BC] SFR_AB12 (-49) *`
And it returns about 16 networks near me which is about what my phone sees also.

[ESP410] returns
```
;Start Scan
;SFR_AB_12	100%	Secure
;End Scan
```

atsju, 2022-06-06:
I also tried to connect using my phone as AP but this doesn't work either. Same error

atsju, 2022-06-06:
Yes I renamed but SFR_AB12 is my network.

atsju, 2022-06-06:
First good news of the day. wifiClientBasic works with Arduino IDE and ESP32 v1.0.6 ...
PlatformIO is supposed to check versions of the libraries so these should not be the problem but it's not the first time for me something broke with it.

atsju, 2022-06-06:
@luc-github as the example worked with Arduino IDE, I did following things: 
- uninstall autobuild marlin (not reinstalled)
- uninstall platformIO
- deleted .platformIO folder to delete everything related to the previous installation
- Re-install platformIO
- git clean -dxf
- Try to build ESP32CAM

This is the outcome :
```
> Executing task: C:\Users\Julien\.platformio\penv\Scripts\platformio.exe run --environment esp32cam <

Processing esp32cam (platform: espressif32@4.4.0; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Check if need to add some library to path
['C:\\GIT\\ESP3D\\libraries', '$PROJECT_LIBDEPS_DIR\\$PIOENV', 'C:\\Users\\Julien\\.platformio\\lib']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Converting esp3d.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WebServer @ 2.0.0      
|   |-- WiFi @ 2.0.0       
|   |-- FS @ 2.0.0
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- Preferences @ 2.0.0    
|-- BluetoothSerial @ 2.0.0
|-- Ticker @ 2.0.0
|-- Wire @ 2.0.0
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.3.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- TFT_eSPI @ 2.4.61
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SPI @ 2.0.0
|-- Ethernet @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- SPI @ 2.0.0
|-- FFat @ 2.0.0
|   |-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- SD_MMC @ 2.0.0
|   |-- FS @ 2.0.0
|-- ESP32SSDP @ 1.2.0
|   |-- ESP32 Async UDP @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- Update @ 2.0.0
|-- ESP8266-Arduino-Lua @ 0.0.30
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- DNSServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- BMx280MI @ 1.2.0
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|-- DHT sensor library for ESPx @ 1.0.6
|-- WebSockets @ 2.3.6
|   |-- Ethernet @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
Building in release mode
Compiling .pioenvs\esp32cam\src\esp3d.ino.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\benchmark.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\debug_esp3d.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\esp3d.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\esp3doutput.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP0.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP100.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP101.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP102.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP103.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP104.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP105.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP106.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP107.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\espcmd\ESP108.cpp.o
In file included from C:\Users\Julien\AppData\Local\Temp\tmp11f7f5ry:1:
C:/Users/Julien/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:209:10: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://registry.platformio.org/search?q=header:pins_arduino.h
*
**********************************************************************

 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\esp32cam\src\esp3d.ino.cpp.o] Error 1
In file included from esp3d/src/core/../include/esp3d_config.h:23,
                 from esp3d/src/core/commands.cpp:21:
C:/Users/Julien/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:209:10: fatal error: pins_arduino.h: No such file or directory
```

I'm now totally unable to build v3.0-alpha2 this would explain why some things weren't lined up correctly on my other builds. It may have used wrong dependencies. However I don't know why it's not building now

luc-github, 2022-06-07:
PIO is great when working but a pain when not 
I did a clean git and delete .platformio  and still can compile, sorry I did not uninstall vscode I am to lazy to reinstall all extensions, so your issue is related to vscode or specific extension IMHO

atsju, 2022-06-07:
Thanks for feedback. I did not uninstall vscode, only platformIO.
Today morning it builds again but still not connecting as client. Really strange, I changed nothing (only restarted computer I suppose).

I tried disabling all extensions. but same problem. Could you please build ESP32CAM and send me .elf and .map output for precisely V3.0-alpha2 ? I might be able to figure out what's different on my side.

luc-github, 2022-06-07:
Did you compiled and flash wificlient sample using using Arduino IDE and Core 2.0.3 ? as it succeed with 1.0.6 I should be interresting to see what happen

atsju, 2022-06-07:
> wifiClientBasic works with Arduino IDE and ESP32 v1.0.6

as tried to explain I used full Arduino IDE build. ESP32 board manager was in version v1.0.6. I don't remember version of ESP8266 but not sure it's relevant ?
What do you mean by "core 2.0.3" ?

luc-github, 2022-06-07:
1.0.6 is old one 
ESP32  v2.0.3 is latest one

atsju, 2022-06-07:
Then I must check the link I used to fetch ESP32 because 1.0.6 was last version I had.
I already noticed different websites share different links. Could you please share the link to add to Arduino IDE "Additional Board manager URL"
Wiki only provides link for ESP8266. I might update this also...

luc-github, 2022-06-07:
in read me of each repo there are links on how to install : 
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html
https://github.com/esp8266/Arduino#installing-with-boards-manager

atsju, 2022-06-07:
Thanks I will have a look tonight

luc-github, 2022-06-07:
I would sugest to just give link to the readme  as the json path may change as it did already

atsju, 2022-06-07:
Agreed. I will update (later).

[14 more replies]

closed 2022-10-18 by luc-github