luc-github/ESP3D #738: [BUG]release 2.1.1 doesn't build with arduino ide and latest ESP8266 (v3.0.2)

atsju, 2022-04-18:
**Describe the bug**
Build failes

**To Reproduce**
Steps to reproduce the behavior:
1. Arduino IDE 1.8.19 with ESP8266 v3.0.2
2. Build release 2.1.1without local modification
3. build failes

**Expected behavior**
Successful build

**Screenshots**

```
notifications_service.cpp:38:10: fatal error: WiFiClientSecureAxTLS.h: No such file or directory
   38 | #include "WiFiClientSecureAxTLS.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
WiFiClientSecureAxTLS.h: No such file or directory
```

**ESP3D Firmware:**
- ESP core version: None
- Version: 2.1.1
- Web UI Version: the one going with 2.1.1
- Wifi mode: not relevant
- Flash method: not relevant
- Any change done in FW: NONE

**Target Firmware:**
not relevant

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: NodeMCU
 - Flash size: 4M

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2022-04-18:
I think it is clear :
[image]

atsju, 2022-04-18:
Sorry, I didn't mean to offense you. Of course this is how I have finally been able to compile and it's great to have the working version list.
However I consider issues as a way of tracking and also that we should keep package up to date when possible to avoid maintenance issues in long term. I have not digged far enough for the moment to understand/judge anything. If the latest development version works with latest ESP8266 version then feel free to just close issue. If it's not the case I would keep it as change request with low priority.

Side note: I'm professional embedded developer and have some experience with other open source repos. This is not a promise but I'm willing to help a bit. You seem motivated and answer fast, this keeps me motivated also.

luc-github, 2022-04-18:
as stated in same screenshot latest esp8266 core and esp32 core are supported by devt version
your question just emphasis what I wrote in your remark about documentation ^_^

atsju, 2022-04-18:
I have seen this while writing the answer. I wanted a confirmation but you are right :) 

Thanks for fast answer

closed 2022-04-18 by atsju