luc-github/ESP3D #739: [Question] How to reset AP connection?

raphaelbs, 2022-04-18:
Hey, first of all, kudos for the work done here!
This project is amazing and I want to contribute to it, especially by supporting the ESP32-Cam!

My question:

I flashed the latest v3.0 into an ESP32-Cam, and after setting up my SSID and static IP, I can't find the ESP in the network anymore. And it also doesn't create the AP.

Tried to reflash it with the v3.0 firmware but had no luck.

What is the recommended approach?

atsju, 2022-04-18:
Hi, 
I'm new to this repo, but I would try erasing flash like recommended at end of [wiki here](https://github.com/luc-github/ESP3D/wiki/Install-Instructions)

raphaelbs, 2022-04-18:
@atsju wow, I didn't see that 😅 
Thank you!

closed 2022-04-18 by raphaelbs