luc-github/ESP3D #740: [Question] no wifi network on nodeMCU

atsju, 2022-04-18:
HW : [cheap 4M nodeMCU](https://fr.aliexpress.com/item/1005002006567767.html?gatewayAdapt=glo2fra&spm=a2g0o.order_list.0.0.49de5e5b6zio9t)
FW: 2.1.1 
Arduino IDE : 1.8.19 + ESP8266 v2.5.2

I flashed the board and after 8seconds I see in serial monitor:
```
M117 ESP3D
M117 192.168.0.1
```
But I see no ESP32 wifi network with my phone.

Following recommendation of erasing whole flash I tried this option in Arduino IDE:
[image]

I believe some bootloader is gone because I sometimes need to press the "flash" button to connect and reflash and this wasn't the case before. (I have several boards, so I will keep one with and one without until I understand everything).
Now this board does show nothing on serial monitor and there is still no wifi network.

Any idea about the bootloader or what settings have been erased ?
Any idea why I do not see wifi network on my phone ?

luc-github, 2022-04-18:
This is not related to bootloader - it is due to quality of the board which do not pull down the GPIO 0 properly and do not go to download mode - it happen sometime on cheap board - esp8266 / esp32 , adding some capacitor  / resistor solve the issue like described here : https://arduino.stackexchange.com/questions/62103/cant-flash-esp8266-anymore-nodemcu
but the common solution is to hold FLASH then rst button before flashing

about you do not see ESP3D network on phone can be due to :
1 - poor antena on esp8266
2 - your phone need time to refresh the list
3 - your phone settings do not accept the chanel of the SSID defined by esp8266 AP because not supported by the country, I saw  german people reporting this, try to change chanel :https://github.com/luc-github/ESP3D/blob/2.1.x/esp3d/config.h#L382
and reflash with erase all flash contents - or do a [ESP444]RESET after 10s boot

In all case check to troubleshoot i SUGGEST try the basic AP example : https://github.com/esp8266/Arduino/blob/2.5.2/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

atsju, 2022-04-19:
Thank you for great feedback.

The AP example works on channel 1 (default) and 11 (as in ESP3D)
The ESP 3D code doesn't work on channel 11 and 1
I also tried `[ESP444]RESET` just in case there were some magic remaining settings somewhere but this didn't help.

luc-github, 2022-04-19:
I do not understand your feedback:
the sample code :https://github.com/esp8266/Arduino/blob/2.5.2/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

does not let you change/set the channel.

So how come  ?
>The AP example works on channel 1 (default) and 11 (as in ESP3D)

atsju, 2022-04-20:
Did I mention I'm a professional embedded dev? And I really try to RTFM a lot before asking ;) 

I just added channel as third parameter. https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html

`  WiFi.softAP(ssid, password);`
I even validated that channel really changes using https://play.google.com/store/apps/details?id=com.farproc.wifi.analyzer

luc-github, 2022-04-20:
well great but you did not mention it - so how I could know what you are doing ?

because my questions to help you are based on what you said you do,  and my debugging test also.

Now how I know what you `really`  did ? may be you did another changes in ESP3D without telling me, because you think it is not important  - which make the debugging not going to the right direction ...

as professional dev you should know information for debugging is important - I am at your side watching what you are doing and I am not mind reader sorry

That means back to square one...

what change did you do in original ESP3D FW to configure - or thinking you are configuring it ?
did this happen with fresh download of FW without any change at all ?

atsju, 2022-04-20:
Right. Worst part is I would have said quite same as you just did...

I use clean git checkout on ((v2.1.1)) without modification (except the channel test).

Just in case, here are my "non code" settings
[image]

luc-github, 2022-04-20:
In that case I only can think about the PHY mode
try commenting 
https://github.com/luc-github/ESP3D/blob/2.1.x/esp3d/wificonf.cpp#L359

and also please do a restart after doing `[ESP444]RESET` this command will reset EEPROM settings but not apply them in live, need a restart to use them

atsju, 2022-04-20:
Tried to comment https://github.com/luc-github/ESP3D/blob/3ff58b07864bd74ec3a57a232546e11d63db2b47/esp3d/wificonf.cpp#L356
plus `[ESP444]RESET` followed by HW reset. => doesn't work.

also commented (in addition) https://github.com/luc-github/ESP3D/blob/3ff58b07864bd74ec3a57a232546e11d63db2b47/esp3d/wificonf.cpp#L467
plus `[ESP444]RESET` followed by HW reset.  => doesn't work.

luc-github, 2022-04-20:
my last idea is to remove all code to be same as sample code:
https://github.com/luc-github/ESP3D/blob/2.1.x/esp3d/wificonf.cpp#L352-L414
to see if some settings are not supported by your phone 
By the way what phone it is ? OS and Version ?

atsju, 2022-04-20:
I will try the code later. If you could point out what I need to change on tag v2.1.1 instead branch 2.1.x it would be great but I may figure it out myself.

Phone Xiaomi Redmi note 7 part number : M1901F7G
Android 10 QKQ1.190910.002
MIUI 12.5.3.0

luc-github, 2022-04-20:
branch 2.1.x is the branch of the tag 2.1.1
[image]

atsju, 2022-04-20:
Do you want me to try on 2.1.x ? or on v2.1.1 ? because the code is not the "exactly" same. As I told you, I use v2.1.1 which is the reference release.

luc-github, 2022-04-20:
I think better try latest git indeed - so branch 2.1.x

atsju, 2022-04-21:
I tried 1e093094cbb63b5eabc660eb9ea78ca4defca56c, made sure to replace/update the libraries in Arduino installation.
I have seen the network but it seemed to be unstable and have been unable to connect.
Tried to change to channel 1 (even if we know channel are not an issue) => same problem
tried https://github.com/luc-github/ESP3D/issues/740#issuecomment-1103492173
=> not working
Tried https://github.com/luc-github/ESP3D/issues/740#issuecomment-1103500505
=> not working

always did `[ESP444]RESET` followed by HW reset (even if I already erase while flash)

retried the AP example https://github.com/esp8266/Arduino/blob/2.5.2/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino
=>working

Retried 1e093094cbb63b5eabc660eb9ea78ca4defca56c but with 3M SPIFF instead 1M (please kindly confirm if this could be issue because I did not retry everything else with these settings)
=> not  working
executed `[ESP710]FORMAT`
=> not working. Note, during formatting, all other wifi except one on ch11 disappeared Radio noise...

What I usually see is that the ESP wifi appears for maybe 1s but then disappears before any connection is done.

luc-github, 2022-04-22:
https://github.com/luc-github/ESP3D/commit/1e093094cbb63b5eabc660eb9ea78ca4defca56c ihas no impact on FW at all - and even no impact on embedded page as it is already built and also no impact after built
https://github.com/luc-github/ESP3D/issues/740#issuecomment-1103492173 make the same behaviour as sample test - if it is not working I have no idea - as this code is now working for almost 2 years

For ESP3D the best on 4MB NodeMCU is 1MB for OTA 1M for Program and 2MB for FS

During Formating the MCU is focus on formating so may be it does not broadcast the signal, that said AP should still be in cache of the phone and not disapear so quickly

atsju, 2022-04-22:
Thank you for update. I may try again later this week.

Are you from France ? If so, might I send you a board physically (for free and you can keep it) ? Not to force you looking at it but I must be missing something dumb or it must be related to the board.
If so shoot me an email (see PR commits messages).

luc-github, 2022-04-22:
Yes I am - but not living there as mentioned in my profile ^_^
[image]

Ok, just for safety I did a check just right now (because who know? I always doubt myself that may missed something...) :

I use this nodeMCU :
[image]
[image]

compiled/flashed :
[image]

So far after 10s on my android phone V11:
[image]
[image]

and on my PC windows 11:
[image]

So for me it is likely some incompatibility with xiaomi android version but I do not know why sample work when same code does not work (the stripped version) on ESP3D on your side

atsju, 2022-04-22:
Link from the board I use is here https://github.com/luc-github/ESP3D/issues/740#issue-1207196280
It's a cheap board (very) poorly build.

This WE I will try with a computer that has wifi and with a a different phone.

luc-github, 2022-04-22:
Yes I saw : no EMI shield like on other node MCU
in all case AP mode is only for configuration - if it is not working you still can move to next step and connect your NodeMCU to your Home AP using direct command from arduino terminal :
https://raw.githubusercontent.com/wiki/luc-github/ESP3D/docs/Commands2_1.txt

atsju, 2022-04-23:
Definitely not working on windows. Tried all 3 board (1 has never been touched). 
I did not try to connect directly with terminal for the moment. 
I will buy other boards to test with. Would [ESP32CAM](https://github.com/luc-github/ESP3D/wiki/ESP-32-CAM ) let me see my printing while also using the wifi to send file to printer ?

luc-github, 2022-04-23:
Well, antenna issue would explain your problems but not why it is always ok with sample code - so the board has more issue I think, and to be honest : you get what you pay for ...
ESP32Cam is supported in ESP3D V3 and webUI V3 
https://github.com/luc-github/ESP3D-WEBUI/issues/123#issuecomment-718693756
In refactoring versions which will replace the V3 soon
https://github.com/luc-github/ESP3D-WEBUI/issues/157#issuecomment-904062902

atsju, 2022-04-23:
well in the end I have 4 boards and paid 0. I knew what to expect, but did not expect it to be this bad.
Thanks for feedback. I will see if rework the board like in forum comment makes it work and might buy ESP32cam.

For now I suppose we can close the issue.

closed 2022-04-23 by atsju