luc-github/ESP3D #741: [V3 Test] Configuration Matrix

luc-github, 2022-04-23:
Here my testing configurations matrix for reference, any other hardware feel free to feedback I can add :
I will add features tested after the first test, also I have ordered some ESP32 with ESP01 form factor, I will add them in test list also when received.

Printer Board | Firmware | ESP | Status | Notes 
---|---|---|---|---
BTT SKR PRO V1.2 | Marlin 2.0.9.3 | ESP01 | |
BTT SKR V1.3 | Marlin 2.0.9.3| Panucatt Wifi Backpack | |With GLCD 128X64
Ramp/Mega | Repetier |  Panucatt Wifi Backpack |  |With GLCD 128X64
MKS Robin Nano V2.0 | Marlin 2.0.9.3 | ESP12 | | MKS Serial Protocol
FYSETC Cheeta V2.0 | Marlin 2.0.9.3 | ESP32Cam | |
Panucatt Aztech X5 Mini | Smoothieware | Panucatt Wifi Backpack | |
AZSMZ mini 2.1 (2015) | Smoothieware | ESP01 | |
Re-ARM /Ramps | Smoothieware | ESP32 | | 
 

TFT are used with Marlin Flashed boards   
TFT| Firmware | ESP | Status | Notes
---|---|---|---|---
BTT TFT35-E3 V3 | | ESP01 | |
BTT TFT70 V3 | | ESP01 | |
MKS TFT32L32|| ESP12| | MKS Serial Protocol
MKS TFT70 V1.1| |ESP12| | MKS Serial Protocol

luc-github, 2022-04-23:
Not just need to test - it will be a long week ...

luc-github, 2022-05-02:
Ok My Azteeg Mini is dead - the SD card reader was unsoldered and I destroy all trying to fix it - too small for my fat toes... 😿 
So I remove it from matrix

luc-github, 2022-05-02:
I would like to test more repetier e.g V2 but I currently have not supported board in hand: https://github.com/repetier/Repetier-Firmware/tree/dev2

luc-github, 2022-05-06:
I did not noticed the Cheetah board need a 24V power supply - which I do not have currently - I will go to buy one next week
My MKS TFT32L32 is dead (Murphy Law) - I have ordered an MKS 35 - so test will be updated

luc-github, 2022-05-14:
For some reason my Cheetah board is actually not functionnal so I need to switch corresponding tests to another one, also I did not put  ESP32-S2 (which is mono core like C3) neither ESP32-S3 (dual core like ESP32), so will add one test for each

luc-github, 2022-05-27:
I have ordered a MKS 3.5 to replace the damaged 3.2, and a BTT Octopus 1.1 to replace the cheetah that is not working.

I still need to order BTT esp32/esp7 modules for the octopus as I saw the ESP SPI pins are connected to Octopus SPI SD so it may me a good oportunity to add direct sd upload on this board => TBC
I will continue testing after the print status is implemented in refactoring webUI

luc-github, 2022-05-30:
I have some delay but still doing tests - ESP32Cam is now working again as well as authentication - I am now working on embedded display which also need a refactoring

A7F, 2023-01-03:
Hi, I managed to install successfully ESP3D on my SKR Mini E3 v1.2 (TMC2209) board by plugging a D1 Mini straight onto the TFT headers. I did not install a TFT touchscreen display anyway, I'm fine with the stock Ender 3 screen, and everything is working perfect: I can provide the wiring diagram if needed, just let me know.

Thank you! :)

luc-github, 2023-01-03:
@A7F you can do PR with all informations to this file https://github.com/luc-github/ESP3D/blob/2.1.x/wiki/hardware-connection.md
it will update the wiki

luc-github, 2023-10-05:
May be add this table result to esp3d.io could be a good reference to complete the hardware section

luc-github, 2024-03-04:
each esp hardware has been added to esp3d.io and almost each printer board are described in how to connect
the board with dedicated connectors will be added later

closed 2024-03-04 by luc-github