luc-github/ESP3D #746: M108 M114 doesn’t work

leonel85, 2022-05-05:
**Describe the bug**
If I send m114 I have no output but it works using usb and prontface 
M108 doesn’t break M0

**To Reproduce**
Steps to reproduce the behavior:
1. Just send m114 or run gcode with M0 and try M108. Nothing happens 

**Expected behavior**
M108 break M0 and M114 get axis coordinates

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- Flash method:  PlatformIO
- Any change done in FW

**Target Firmware:**
 - Name: Marlin
 - Version  2.x
 
**Board used (please complete the following information):**
 - MCU: esp01
 - Name:esp01
 - Flash size: 1M

luc-github, 2022-05-05:
Looks like communication issue on serial port used for ESP3D
what board are you using ?
how did you configure Marlin to communicate with board ?
did you ensure baud rate are same between Marlin and ESP3D ?

leonel85, 2022-05-06:
I have trigorilla v1.1  (anycubic i3 mega X)
I follow some examples on wiki and I use esp01 and logic level converter
Baud rate is 250000
( https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S
#define BAUDRATE 250000   //default value )
I try other GCODES and they works
The baud rate is the same that I use with pronterface through USB.

luc-github, 2022-05-06:
you mean in WEBUI  terminal the GCODE works ?
can you do a screenshot of output of in webUI ?

M115
M105
M114

leonel85, 2022-05-06:
Yes I mean in WEBUI but not M108 and M114 only
G28, move axis and so on

unfortunately I’m not at home and I can test this commands now but when I try M114 I have no output

luc-github, 2022-05-06:
No problem send me the output when you have it

leonel85, 2022-05-06:
Ok
Thanks

leonel85, 2022-05-06:
This is the output
I will send also one screenshot because some characters are not recognized

M115
:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:1
ok
.
ՉPrinter EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:1
Cap:PROMPT_SUPPORT:1
Cap:SDCARD:1
M105
M114

luc-github, 2022-05-06:
M105 is not recognised neither ?
wait - what version of FW and WebUi are you using ?

leonel85, 2022-05-06:
[image]

luc-github, 2022-05-06:
Obviously your serial is messy, the output of M115 is incorrect, it miss first lines  and has a lot of garbage - M105 / M114 does not answer but their answer is short so I am guessing data is corrupted, please use 115200 as baudrate with ESP3D, I suspect your logic shifter cannot handle 2500000

leonel85, 2022-05-06:
To do that I think I have to change  #define BAUDRATE 250000 //default value  to 115200  , recompile Marlin and update

luc-github, 2022-05-06:
and also updated serial baudrate in webUI

leonel85, 2022-05-06:
I will do it when I go back to home and I will tell you if it works.
I think I can do it after tomorrow
Thanks

leonel85, 2022-05-10:
I try to change the baud rate to 115200 and I have the same problem 
With pronterface it works

leonel85, 2022-05-10:
Probterface output
[image]

leonel85, 2022-05-10:
Esp3D output

[image]
[image]

luc-github, 2022-05-10:
Yes it is not software issue but hardware,  with pronterface you do not use the serial with your logic shifter but the USB, so output is correct
remove the logic shifter and should be ok I think - the esp8266 can handle short term 5V

leonel85, 2022-05-10:
Thanks 
I will try without logic shifter

atsju, 2022-05-10:
Hi, I made some research this weekend because I had 5V MCU on 3V3 ESP8266. I confirm ESP can NOT be supplied in 5V and input are not officially 5V tolerant BUT I have read many thread (some of them with high quality electronics test to confirm the supposition) and the inputs are 5V tolerant. 

Conclusion: DO NOT supply ESP with 5V but inputs are 5V tolerant. If you want to be 200% sure, you can just put a serial resistor (say 1k to 10k) between MCU Tx and ESP Rx. This will protect input (I will write some wiki about this next month).

leonel85, 2022-05-10:
Hi, I remove the logic converter and use voltage devider resistor to the RX pin (1k, 2k).
Now M115 is ok and M114 is working.
I have to solder new components but it works with breadboard

luc-github, 2022-05-11:
great

luc-github, 2022-05-11:
I close ticket then

closed 2022-05-11 by luc-github

leonel85, 2022-05-11:
Thanks so much