luc-github/ESP3D #747: [BUG]<ESP3d included in Marlin 2.0.9.3 - No media error>

ul42, 2022-05-05:
Bonjour à vous (désolé, je suis un peu fatigué ;) )

**Describe the bug**
Mon "plutôt handicapant" problème est que la carte Sd n'est absolument pas detectée.
Ni par Marlin, ni par ESP3D.

**To Reproduce**
Steps to reproduce the behavior:
Quand je clic sur "rafraichir", l'interface "SD" m'affiche un fichier nommé "OK" et qui ne sert apparament absolument a rien, et l'invite de commande me marque : "
echo:No media
ok
". Toute les actions depuis l'interface web me donne ces deux erreurs ainsi qu'un bug faisant reboot l'esp après de l'upload d'un gcode nommé "TstCube.gco".

**Expected behavior**
Je m'attend personellement a ce qu'il y ait un affichage, ainsi que pas de bug a l'upload d'un Gcode.

**Screenshots**
![ESP3d Problem 1](https://user-images.githubusercontent.com/85800590/166992562-4cf2a3dd-bfa5-45dd-b29e-5cc378e56501.PNG

**ESP3D Firmware:**
- ESP core version: non connue
- Version: non connue
- Web UI Version: 2.1b77
- Wifi mode: AP
- Flash method: PlatformIO
- Any change done in FW : Demander moi pour avoir les fichiers de config.
 
 tips: to get ESP3D informations type `[ESP420]plain`    
       to get WebUI version check your menu header    

**Target Firmware:**
 - Name: Marlin
 - Version: 2.0.9.3
 
 tip: to get firmware info type `M115` in console    
J'ai envoyé plusieures commandes tels que :

[ESP800]
FW version:2.0.9.3-1.0.11 # FW target:marlin-embedded  # FW HW:No SD # primary sd:/sd # secondary sd:none # authentication:no # webcommunication: Sync: 81:192.168.0.1# hostname:marlinesp(AP mode)

[ESP420]
Chip ID: 5152
CPU Frequency: 240Mhz
CPU Temperature: 71.1C
Free memory: 209.25 KB
SDK: v3.2.3-14-gd3e562907
Flash Size: 4.00 MB
Available Size for update: 1.81 MB
Available Size for SPIFFS: 287.03 KB
Baud rate: 250000
Sleep mode: None
Web port: 80
Hostname: marlinesp
Active Mode: AP (30:C6:F7:05:20:15)
SSID: MARLIN_ESP
Visible: Yes
Authentication: WPA2
Max Connections: 4
DHCP Server: Started
IP: 192.168.0.1
Gateway: 0.0.0.0
Mask: 255.255.255.0
Connected clients: 1
34:6F:24:A3:E7:31 192.168.0.2
Disabled Mode: STA (30:C6:F7:05:20:14)
FW version: Marlin 2.0.9.3 (ESP3D:1.0.11)

**Board used (please complete the following information):**
 - MCU: ESP32S
 - Name: Similaire a une Fysetc E4
 - Flash size: 4M

**Additional context**
Je suis en train de concevoir une machine FDM controllé par un ESP32S de Ai-Thinker ;)
J'en viens à vous surtout par manque de temps.

J'ai essayer plusieurs solutions :
- reformaté la carte SD.
- vérifié les paramètres de ESP3d, ESP3d WebUI, et du FW.
- activé/désactivé certaines options dans le marlin (voir les fichiers de config).
- vérifié le noms des gcodes contenu dans la sd.
- vérifié les schematics des cartes que j'utilise (les design sont propriétaire a moi même).
- ... (J'ai du en oublié d'autre).

Je pense que le faite est que je ne suis pas encore bien familliarisé avec l'environement logicielle des ESP32 et de ESP3D. Pour Marlin3d, je ne suis pas a ma première machine.

Je vous remercie d'avance pour l'aide.
Cordialement.

ul42, 2022-05-05:
Voici les fichiers de config "Configuration_adv.h" et "Configuration.h" :
[ref.zip](https://github.com/luc-github/ESP3D/files/8633826/ref.zip)

luc-github, 2022-05-05:
En utilisant le port USB connecter a un PC et un terminal serie ou pronterface our repetier host
qu'est que les commandes :
`M21` 
`M20` 
renvoient ?