luc-github/ESP3D #749: [BUG]missing index.html.gz

atsju, 2022-05-07:
**Describe the bug**
[index.html.gz](https://github.com/luc-github/ESP3D/blob/v2.1.1/esp3d/data/index.html.gz) is nowhere in 2.1.x anymore

**To Reproduce**
File is missing in repo

**Expected behavior**
A beginner should be able to understand what to upload in less than 1h. I'm lost :( just needed to vent
Mention #737

**Screenshots**
none

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:
- Wifi mode: AP (not finished)
- Flash method: Arduino Ide
- Any change done in FW
- NONE

 
 tips: to get ESP3D informations type `[ESP420]plain`    
       to get WebUI version check your menu header    

**Please, what is ESP core version VW version ? Do you want a version + a sha ? 
could your bug template and answers from [ESP420]plain be in same format to help you get good filling ?**

**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 2.0]
 
 tip: to get firmware info type `M115` in console    

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name:
 - Flash size: 4M

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

atsju, 2022-05-08:
I understand now this is generated from webUI repo. I suppose it could be better documented for development branch and to explain to beginners but doc is correct for the release.

luc-github, 2022-05-11:
I removed because was outdated and forget to put english index.html.gz instead as latest version multilangage version is too big for some partition,

I will update today

luc-github, 2022-05-11:
done

closed 2022-05-27 by luc-github