luc-github/ESP3D #750: [FEATURE REQUEST] bug template enhancement

atsju, 2022-05-07:
**Is your feature request related to a problem? Please describe.**
While opening a bug I tried to get requested infos but it has been unclear for me (as a new user) what  was neededd

**Describe the solution you'd like**

this is the bug template
```
**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- Flash method: [Arduino Ide / PlatformIO]
- Any change done in FW
 
 tips: to get ESP3D informations type `[ESP420]plain`    
       to get WebUI version check your menu header    
```

- move tips up as it will be read before the filling 
- Use the format `` to show lines during template filling that will not show in rendering
- make template items correspond to what `[ESP420]plain` returns

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

luc-github, 2022-05-27:
@atsju can this issue be closed as all PR are now merged ?

atsju, 2022-05-27:
Please could you modify the template as I mentionned in #751 ? I recommend using same wording/order/info in version requested by github as it the one given by serial of ESP (and/or add some comme t like "can be found at top right of web page"). I remember opening my first issue here I was a bit confused. 
I think it could help users but I don't have the knowledge to do it. For you it will be 10 minutes work.

luc-github, 2022-05-27:
do you mean 
 feature request template using :
 ```
ESP core version: [e.g. 2.5.2]
Version:[e.g. 2.1b4]
Web UI Version:[e.g. 2.1b14]
Wifi mode:[e.g. STA/AP]
Flash method: [Arduino Ide / PlatformIO]
Any change done in FW:
```

atsju, 2022-05-27:
yes.
I know it might be clear for you (and it becomes each day clearer for me) but as a new user I was very confused.

atsju, 2022-05-27:
sorry if I haven't been clear. Could you modify this file :https://github.com/luc-github/ESP3D/blob/2.1.x/.github/ISSUE_TEMPLATE/bug-report---problem.md

it's the bug report file, not the feature request. The information are needed but unclear (for new users)

luc-github, 2022-06-01:
Hi Sorry what is unclear ?  - it mostly ask for version ...
the only possible issue could be 
`ESP core version: [e.g. 2.5.2]`
because this is for Arduino IDE and infrmation is in Board manager, for platformio if they use the provide env is not necessary if they provide the version of ESP3D FW they use

PR #774 (2022-06-02): more explaination about requested parameters
close  #750 

@luc-github With the additional info from your comment it was now clearer and I made the changes I thing should suffice.
You know your code way too much to understand how difficult it is for someone new here to understand what to do.
closed 2022-06-02 by luc-github