luc-github/ESP3D #753: [FEATURE REQUEST] security by default

atsju, 2022-05-08:
**Is your feature request related to a problem? Please describe.**

All user are not security aware.
IMO `#define AUTHENTICATION_FEATURE` should be set by default.

**Describe the solution you'd like**

set `#define AUTHENTICATION_FEATURE` by default and user can deactivate it

**Describe alternatives you've considered**

I understand this can be inconvenient for users that are only on local network. Maybe this could be configured at first connection (again, proposing WITH security as default choice)

**Additional context**

At this point I'm only planning on opening a port on my internet box and reviewing security level of connected device. Maybe I don't have all information at this point

luc-github, 2022-05-08:
Authentication is not an 'option' to avoid to be disabled, so it is a define, because all function are hard coded and not optional.

Password is never sent to webui it actually send *********, so been in clear is not a problem, that said using password box it is already implemented in 3.x, same for multiple setting and one save button, same for restart message if setting need restart.

I am not in favor of setting authentication enable by default,  because it is not a setting to disable it but need reflash, also it will bring me too much sustaining work, people which open their network to outside should be aware of what they are doing and so check security feature.

luc-github, 2022-05-08:
be noted there is no https support yet

atsju, 2022-05-08:
Thanks for feedback. Great to see many of the little usage problems in 2.x have already been fixed in 3.x.

I fully understand autentication will bring more work. I still maintain my position but this may be done (much) later if real issues are discovered about it.

Feel free to close this issue if you think everything is addressed. I might open new issue in webUI usage when I try 3.x and all latest features.

luc-github, 2022-05-27:
I close issue then

closed 2022-05-27 by luc-github