luc-github/ESP3D #754: SD file Upload error

leonel85, 2022-05-12:
**Describe the bug**

When I try to upload a file on the SD I have an format error 8.3.
Then I try to load a file called TEST.GCO. I don't have the same error but the file is not on the sd
All files in the sd as extension .gcode

**To Reproduce**

upload file 

**Expected behavior**

file is not loaded in the sd card

**Screenshots**

[image]

**ESP3D Firmware:**

[ESP420]plain
Chip ID: 4055564
CPU Frequency: 160Mhz
Free memory: 38.52 KB
SDK: 2.2.1(cfd48f3)
Flash Size: 1.00 MB
Available Size for update: 376.00 KB(Not enough)
Available Size for SPIFFS: 228.20 KB
Baud rate: 115200
Sleep mode: Modem
Channel: 2
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: 3D_Printer_01
Connected to: qwerty
Signal: 68%
IP Mode: DHCP
Captive portal: Enabled
SSDP: Enabled
NetBios: Disabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Marlin
M117 output: Enabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.1.b8 ESP8266/8586

- Wifi mode:[e.g. STA/AP]
- Flash method: PlatformIO
- Any change done in FW: baud rate 115200, comment //#define NOTIFICATION_FEATURE

**Target Firmware:**

M115
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
FIRMWARE_NAME:Marlin 2.0.x (knutwurst, Github) (May  9 2022 00:55:22) SOURCE_CODE_URL:github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic Printer EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:1
Cap:PROMPT_SUPPORT:1
Cap:SDCARD:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:1
ok

**Board used (please complete the following information):**

- MCU:esp01
- Name:esp01
- Flash size: 1M

luc-github, 2022-05-13:
if the progress bar is still visible, it means : upload is not finalized, it can take a long time because protocol is very very slow
 the 8.3 message is not an error : https://github.com/luc-github/ESP3D/discussions/326
 
 if you do not have an error message when upload is done or aborted - it just mean you did not wait enough time and refresh the page which may abort the upload
 
just in case to  be sure:
1 -  you must use 4GB sd card as Marlin does not know how to write, on higher sd card, he can read but not write, 
2 - be sure your sd is formated with https://www.sdcard.org/downloads/formatter/, never use your OS to format an SD card
3 - take time to read : https://github.com/luc-github/ESP3D/discussions/categories/f-a-q

luc-github, 2022-05-19:
@leonel85 any update ?

leonel85, 2022-05-19:
Hi
1 - I'm using an 4GB SD Card
2 - I format the sd is with https://www.sdcard.org/downloads/formatter/
Now I'm able to upload files with format 8.3 but my printer show only files with extension .gcode so I can't use uploaded files from the printer.
I try to remove the limit 8.3 on index.html. I have no message for the format but write error.

I'm not able to use files and I don't know what to do.

luc-github, 2022-05-20:
yes I told you Marlin do not handle long file name for upload - removing the message will not solve the issue, 
contact the Marlin fork owner - knutwurst to ask him why its FW limit to .gcode file the file display

luc-github, 2022-05-20:
So if upload is now working I close issue  -  Marlin part is out of scope

closed 2022-05-20 by luc-github

leonel85, 2022-05-29:
Marlin add LONG_FILENAME_WRITE_SUPPORT I think from version 2.0.6.3
Knutwrust didn't answer me about format limitation so I modify source code by my self.
Now it works. Can you remove 8.3 limitation since it has been solved by Marlin ?

At moment I remove file limitation from Knutwrust Marlin and ESP3D and it works

Thanks a lot

luc-github, 2022-05-29:
PR has been done in January this year : https://github.com/MarlinFirmware/Marlin/pull/23526
last release 2.0.9.3 has been done in December last year - so this is not in any release yet - so it wont't be removed in short term,  as not everyone will do update right away

leonel85, 2022-05-29:
You are right. 
To fix it I use  [bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)
 It works and I think this will be used for next releases and some one else use it.
 In my case I fix only scard features. It could be that some one use bug fix-2.0.x and can use ESP3D without limitation.
 Anyway for me it's working but if some one else has the same problem "can fix it" using bug fix-2.0.x or new releases.

 Thanks, this is the only issue that help me to understand the problem and fix it. Now I'm happy to use ESP3D also for file management.

luc-github, 2022-06-05:
@leonel85 additionnaly as I forget to mention it: the long name support is still buggy per my tests, it does not works if you have subdirectories 
https://github.com/luc-github/ESP3D-WEBUI/issues/221#issuecomment-1107824823