luc-github/ESP3D #755: [Question]<Can ESP32S2 support the camera of ov2640>

liqijian101, 2022-05-14:
What is your question ?
hi luc, I test the esp32s2 chip with esp-12 . It could not work . Did you test that?
thanks your .

luc-github, 2022-05-14:
S2 is supported in refactoring branch which is not yet merged to V3
but I do not have S2 with camera, so I cannot says if it works, is there some devices available already ?

liqijian101, 2022-05-14:
I designed an S2 module based on esp32Cam. Because I need more pins, is esp3d refitting working, like uploading files.

liqijian101, 2022-05-14:
I designed an S2 module based on esp32Cam. Because I need more pins, is esp3d refactoring working, like uploading files.

luc-github, 2022-05-14:
here latest tests
https://github.com/luc-github/ESP3D/issues/741

I did quick check on S2 and S3 but no camera on them

liqijian101, 2022-05-14:
Another reason I tested esp32s2 was to add the ability to connect to the CH340 serial chip

liqijian101, 2022-05-14:
If I use a Webcam, is the following setting necessary.
-DBOARD_HAS_PSRAM
-mfix-esp32-psram-cache-issue

luc-github, 2022-05-14:
Use PSRAM with camera is recommended, so if you added PSRAM yes this is necessary, I did not check if S2 has different setting for camera as I do not have board, I am guessing they are identical to ESP32,