luc-github/ESP3D #758: Index.html.gz connection failed

aacorvalan, 2022-05-18:
Hello, in version 3.0 of esp3d when index.html.gz is loaded from https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/languages/es/index.html.gz it indicates connection failed, however in 2.1 it works.
[image]
[image]

aacorvalan, 2022-05-19:
Uploading the version from https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/dist/printer/production/index.html.gz I get...
[image]

aacorvalan, 2022-05-19:
In version 2 preferences.json is created, in version 3 as I did not enter ESP3D it could not be created
[image]

luc-github, 2022-05-19:
yes it is not an issue, it will be created when you will save it

aacorvalan, 2022-05-19:
With version 3, when I upload index I can't access the firmware in any way

luc-github, 2022-05-19:
I will check later today

aacorvalan, 2022-05-19:
I thank you very much for your time. I will try to load it in ESP32-cam to put the SD, since I cannot access SD tft on my Artillery genius pro

luc-github, 2022-05-19:
I suspect FS is corrupted , need format

aacorvalan, 2022-05-19:
I don't remember if this is from v2 or v3, now i print from pendrive for send command

{"chip_id":"7724577","cpu":"160","freemem":"35.84 KB","SDK":"2.2.1(cfd48f3)","flash_size":"4.00 MB","update_size":"534.09 KB","spiffs_size":"1.86 MB","baud_rate":"115200","sleep_mode":"Modem","channel":"1","phy_mode":"11n","web_port":"80","data_port":"8888","hostname":"esp3d","active_mode":"STA (80:7D:3A:75:DE:21)","connected_ssid":"Mi router","connected_signal":"100%","ip_mode":"DHCP","ip":"192.168.1.111","gw":"192.168.1.1","msk":"255.255.255.0","dns":"192.168.5.1","disabled_mode":"AP (82:7D:3A:75:DE:21)","captive_portal":"Enabled","ssdp":"Enabled","netbios":"Disabled","mdns":"Enabled","web_update":"Enabled","pin recovery":"Disabled","autentication":"Disabled","target_fw":"Marlin","M117_output":"Enabled","Notifications":"Disabled","Serial_output":"Enabled","Websocket_output":"Enabled","TCP_output":"Enabled","fw":"2.1.2 ESP8266/8586"}

aacorvalan, 2022-05-19:
In esp32-cam it works, I don't know why in nodemcu it doesn't
[image]

luc-github, 2022-05-19:
did you configure the FW to support SD card of the ESp32cam ? - it is an SDIO card
But what do you want to do with this SD card ?
Esp32cam SD card cannot be seen by printer, so any file uploaded on it, won't be accessible for Printer

About esp8266 I suspect a FS corruption and you need to erase whole card or use the terminal and type [ESP710]FORMAT and restart bord when format is done

aacorvalan, 2022-05-19:
I need to load GCO file and print from ESP3d to not use pendrive or SD in TFT, also resume printing due to lack of electricity. In SDIO not upload files

aacorvalan, 2022-05-19:
Is it possible to load Gco or Gcode to Esp-cam memory and print from Esp3d?

luc-github, 2022-05-20:
ESP3D does not stream GCODE file yet

luc-github, 2022-05-27:
I close issue as invalid

closed 2022-05-27 by luc-github