luc-github/ESP3D #761: [Question]<How can I firgue the error position in platform io>

liqijian101, 2022-05-20:
What is your question ?
[ESP3D][camera.cpp:72] handle_snap(): Camera capture ongoing
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x400dcda5  PS      : 0x00060930  A0      : 0x800dc44a  A1      : 0x3ffd4cd0  
A2      : 0x00000fa0  A3      : 0x00015acf  A4      : 0x3f780a8c  A5      : 0xffffd9ff  
A6      : 0x3ffd4c80  A7      : 0x00000008  A8      : 0x3f7846c1  A9      : 0x3f780ac4  
A10     : 0x0000d9ff  A11     : 0x000079d2  A12     : 0x3ffd1f94  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffd4c94  SAR     : 0x00000004  EXCCAUSE: 0x00000009  
EXCVADDR: 0x3f7846c1  LBEG    : 0x3ffd1f94  LEND    : 0x00000000  LCOUNT  : 0x40027eed  

Backtrace:0x400dcda2:0x3ffd4cd00x400dc447:0x3ffd4d00 0x4008cca4:0x3ffd4d20 0x4009378d:0x3ffd4d80 0x401386a5:0x3ffd4dc0 0x400a0346:0x3ffd4de0 0x400a03ed:0x3ffd4e00 0x400a0473:0x3ffd4e40 0x400a060e:0x3ffd4ea0 0x400956aa:0x3ffd4ef0 0x40096348:0x3ffd4f10 0x40096047:0x3ffd4f30 0x4008429a:0x3ffd4f50 0x400833e2:0x3ffd4f70 0x400ac231:0x3ffd4f90 

ELF file SHA256: 0000000000000000

Rebooting...

luc-github, 2022-05-20:
do you have psram ?

liqijian101, 2022-05-20:
yes. I use esp3d v3.0. has 8M psram

luc-github, 2022-05-21:
use stack decoder and the elf of the binary to decode the stack , I usually use arduino ide version of stack decoder, for PIO I do not know sorry, you need to check the  PIO documentation

luc-github, 2022-05-21:
Is that your S2 MCU with camera support ?

liqijian101, 2022-05-21:
yes. s2 mcu

luc-github, 2022-05-21:
so did you modified S2 env  to support camera ( E.g: add PSRAM ?)
remove the `-mfix-esp32-psram-cache-issue` to check

liqijian101, 2022-05-21:
I remove  the -mfix-esp32-psram-cache-issue . it also has errors.

Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x400db6fd  PS      : 0x00060930  A0      : 0x800dada2  A1      : 0x3ffd4c40  
A2      : 0x00000fa0  A3      : 0x000060b6  A4      : 0x3f780a8c  A5      : 0xffffd9ff  
A6      : 0x3ffd4bf0  A7      : 0x00000008  A8      : 0x3f7846c1  A9      : 0x3f780ac4  
A10     : 0x0000d9ff  A11     : 0x0000b606  A12     : 0x3ffd1f94  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffd4c04  SAR     : 0x0000001f  EXCCAUSE: 0x00000009  
EXCVADDR: 0x3f7846c1  LBEG    : 0x3ffd1f94  LEND    : 0x00000000  LCOUNT  : 0x40027eed  

Backtrace:0x400db6fa:0x3ffd4c400x400dad9f:0x3ffd4c70 0x4008cc0c:0x3ffd4c90 0x40093481:0x3ffd4cf0 0x40136e89:0x3ffd4d30 0x4009f83e:0x3ffd4d50 0x4009f8e5:0x3ffd4d70 0x4009f967:0x3ffd4db0 0x4009fafd:0x3ffd4e10 0x40095252:0x3ffd4e60 0x40095e60:0x3ffd4e80 0x40095b7b:0x3ffd4ea0 0x40084366:0x3ffd4ec0 0x400834de:0x3ffd4ee0 0x400aac0d:0x3ffd4f00 

ELF file SHA256: 0000000000000000
Rebooting...

platform = https://gitee.com/li-qijian/platform-espressif32.git#feature/arduino-upstream
board = esp32-s2-saola-1
framework = arduino
; https://github.com/espressif/arduino-esp32.git
platform_packages = framework-arduinoespressif32 @ https://gitee.com/li-qijian/arduino-esp32.git
;board_build.arduino.upstream_packages = no
monitor_speed = 115200
monitor_flags = --echo
monitor_filters = send_on_enter, colorize, esp32_exception_decoder
board_build.f_cpu = 240000000L
board_build.mcu = esp32s2
board_build.variant = esp32s2
; set frequency to 80MHz
board_build.f_flash = 80000000L
board_build.flash_mode = dio
; None
build_flags =   -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0
                -DBOARD_HAS_PSRAM
;                -mfix-esp32-psram-cache-issue
board_build.partitions = min_spiffs.csv
upload_speed = 460800
extra_scripts = pre:platformIO/extra_script.py
lib_ignore = 
    TFT_eSPI

luc-github, 2022-05-21:
well try using arduino ide - I have some strange issue with PIO recently that I cannot reproduce using arduino IDE
I am using latest 4.3.0 and even having issue