luc-github/ESP3D #767: [Question]<Error (SyntaxError: Unexpected token F in JSON at position 0)>

liqijian101, 2022-05-28:
I Use ESP32-S3 and WEBUI 3.0.0.85

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
SHA-256 comparison failed:
Calculated: 0bb45b01207a4a9e9fca8880ddfc0f79112bfa4d004bd1e6b2eaa75c5323b19d
Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
Attempting to boot anyway...
entry 0x403b61d8
;Setup wifi
M117 Setup wifi
;Connecting
M117 Connecting
;192.168.1.185
M117 192.168.1.185
;192.168.1.185
M117 192.168.1.185

[env:esp32-s3]
platform = espressif32@4.3.0
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
monitor_flags = --echo
monitor_filters = send_on_enter, colorize, esp32_exception_decoder
board_build.f_cpu = 240000000L
board_build.mcu = esp32s3
board_build.variant = esp32s3
; set frequency to 80MHz
board_build.f_flash = 80000000L
board_build.flash_mode = qio
;uncomment and modify if board is not 4MB
board_upload.flash_size = 8MB
;board_build.partitions = default_16MB.csv
; None
build_flags =   -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0
                -DBOARD_HAS_PSRAM
board_build.partitions = min_spiffs.csv
upload_speed = 460800
extra_scripts = pre:platformIO/extra_script.py
lib_ignore = 
    TFT_eSPI

luc-github, 2022-05-28:
for S3 the support is present in https://github.com/luc-github/ESP3D/tree/3.0-refactoring
and must use webUI https://github.com/luc-github/ESP3D-WEBUI/tree/3.0-rewrite

webui WEBUI 3.0.0.85 is not the rewrite version pls try https://github.com/luc-github/ESP3D-WEBUI/tree/3.0-rewrite/dist/Printer3D/Marlin

liqijian101, 2022-05-28:
Thanks Luc. 
 I  try  the  [3.0-rewrite]. it can work sucessfully.
Does the  WebUI 3.0-rewrite support camera?

luc-github, 2022-05-28:
Yes it has now same level as 3.0 - it should be the new webui 3.0 early next week -

liqijian101, 2022-05-28:
I am looking forward to using the camera. I have given up esp32 s2 I made another S3 module

luc-github, 2022-05-28:
is that working better with S3 ?

liqijian101, 2022-05-28:
yes  S3 is better

liqijian101, 2022-05-28:
actually I hope that the  webui 3.0 can be used .because I think 3.0 is more beautiful

luc-github, 2022-05-28:
which one ? the 3.0 or the 3.0 refactoring ?

liqijian101, 2022-05-28:
The 3.0.

luc-github, 2022-05-28:
with the 3.0 refactoring you can do your own  theme / color - so you can almost get same I guess

liqijian101, 2022-05-28:
Yes, I’m going to spend some time on 3.0 refactoring. Maybe I just don’t know it well enough to use it

luc-github, 2022-05-28:
check this : https://github.com/luc-github/ESP3D-WEBUI/blob/3.0-rewrite/Features.md

liqijian101, 2022-05-28:
IP Camera & ESP32 Camera display support
I see it already supports the camera, doesn’t it

luc-github, 2022-05-28:
Yes I told you before

liqijian101, 2022-05-28:
But I can't find the esp32-cam setting.
[image]

luc-github, 2022-05-28:
cannot see the picture 
if you enabled esp32cam it should appear in panel settings, this is 3.0 but settings are same with refactoring : https://github.com/luc-github/ESP3D-WEBUI/issues/123#issuecomment-718693756

liqijian101, 2022-05-29:
[image]

I find the camera name is none

luc-github, 2022-05-29:
so it means camera is not properly setup

luc-github, 2022-05-30:
@liqijian101 the camera support was broken in latest webUI - I have fixed it today please give try 
I enclose a screencast I did today to show how to configure it
[image]

liqijian101, 2022-05-30:
Thank you very much. The camera is now working properly. But can the refresh time be set to any value? Now it can only be a multiple of 0 or 100. I want to set it to 10 or 20

luc-github, 2022-05-30:
you can put any value but less than 100ms is not really realistic - this is a web query not direct transmission - I do not think you can have a query + do a snap +transmist the  image in less than 100ms so in 10ms ... ,

Also this keep the esp board busy just doing the camera snap and will not be able to handle any other commands from web or printer - this is not a PC after all- if you want a a full video stream you must forget to use the esp to monitor the printer, and in  this case ESP3D is not the solution.

From another point of view I do not see the purpose to have such high rate for monitoring a CNC or 3dprinter job - usually people do time lapse which mean 1 image per layer and 1 layer can take several seconds, even minuntes, so not a high speed stream.
200ms is in the context of 3D printer monitoring, a good mitigation IMHO