luc-github/ESP3D #775: [BUG] ESP3D 3.0  compilation terminated.

imhmao, 2022-06-05:
**Describe the bug**

In file included from sketch\src\core\benchmark.cpp:21:0:
sketch\src\core\../include/esp3d_config.h:31:32: fatal error: ../esp3dlib_config.h: No such file or directory
 #include "../esp3dlib_config.h"
                                ^
compilation terminated.
exit status 1

**To Reproduce**

**ESP3D Firmware:**

- ESP3D FW version: 3.0
- Flash method: [Arduino Ide ]  
    If you build with Arduino:
    - Arduino ESP8266 package version:  2.7.4
- Any change done in FW:

**Target Firmware:**

- Name:  Marlin
- Version  2.0

**Board used (please complete the following information):**

- MCU:ESP8266 
- Name:Generic ESP8266 Module
- Flash size:  4M

luc-github, 2022-06-05:
What change did you do ? it looks like you deleted configuration.h
[image]

The CI build is passed as well a my Arduino

Additionnaly but this is not rel;ated you your issue :
do not use  ESP8266 package version: 2.7.4 but 3.X or git version as mentioned in readme

[image]

imhmao, 2022-06-05:
Thanks for your reply.  It worked after upgrading my ESP8266 to 3.02.

luc-github, 2022-06-05:
Ok great I close issue then

closed 2022-06-05 by luc-github