luc-github/ESP3D #781: [Question]<PSRAM ID read error>

liqijian101, 2022-06-26:
The chip I use is esp32-s3-wroom-1 n4r8. When I turn on PSRAM, the prompt error is "E (205) psram: PSRAM ID read error: 0x00ffffff" .
And now I can only use USB to download programs. I can't use UART0. UART0 doesn't respond.
Luc, do you have any solutions? I look forward to your reply

[env:esp32-s3]
platform = espressif32@4.4.0
board = esp32-s3-devkitc-1
;board =esp32s3box
framework = arduino
monitor_speed = 115200
monitor_flags = --echo
monitor_filters = send_on_enter, colorize, esp32_exception_decoder
board_build.f_cpu = 240000000L
board_build.mcu = esp32s3
board_build.variant = esp32s3
; set frequency to 80MHz
board_build.f_flash = 80000000L
board_build.flash_mode = qio
;uncomment and modify if board is not 4MB
;board_upload.flash_size = 16MB
;board_build.partitions = default_16MB.csv
; None
build_flags =   -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0  -DCONFIG_IDF_TARGET_ESP32S3=1
                -DBOARD_HAS_PSRAM
board_build.partitions = min_spiffs.csv
extra_scripts = pre:platformIO/extra_script.py
upload_speed = 460800
lib_ignore = 
    TFT_eSPI

luc-github, 2022-06-26:
Does your board has PSRAM ?
does this sketch is working ?
```
#include <Arduino.h>

void setup() {
  log_d("Total heap: %d", ESP.getHeapSize());
  log_d("Free heap: %d", ESP.getFreeHeap());
  log_d("Total PSRAM: %d", ESP.getPsramSize());
  log_d("Free PSRAM: %d", ESP.getFreePsram());
}

void loop() {}
```

liqijian101, 2022-06-26:
that sketch can not work, it has the same error.
My board did have PSRAM.

luc-github, 2022-06-26:
It means the PSRAM is not detected - for that ESP3D cannot do anything 
Check if pins are correct for S3 as pinout change a lot https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/external-ram.html

I suggest to raise question on ESP32 discord or gitter  they may have some clue - as I does not have suche device and so never encounter the issue

liqijian101, 2022-06-26:
I found the cause of the problem. Because the PSRAM on the chip I'm using is the OCTAL SPI, not the Quad SPI. I set up OPI PSRAM in the Arduino ide and it works. But how do I set this  in platform io

luc-github, 2022-06-26:
it seems it is `dout` when I flash with OPI it do   `--flash_mode dout` but it is not enough I think 
I see:
```
esp32s3.menu.PSRAM.disabled=Disabled
esp32s3.menu.PSRAM.disabled.build.defines=
esp32s3.menu.PSRAM.disabled.build.psram_type=qspi
esp32s3.menu.PSRAM.enabled=QSPI PSRAM
esp32s3.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM
esp32s3.menu.PSRAM.enabled.build.psram_type=qspi
esp32s3.menu.PSRAM.`=OPI PSRAM
esp32s3.menu.PSRAM.opi.build.defines=-DBOARD_HAS_PSRAM
esp32s3.menu.PSRAM.opi.build.psram_type=opi
```
and also
```
esp32s3.menu.FlashMode.opi=OPI 80MHz
esp32s3.menu.FlashMode.opi.build.flash_mode=dout
esp32s3.menu.FlashMode.opi.build.boot=opi
esp32s3.menu.FlashMode.opi.build.boot_freq=80m
esp32s3.menu.FlashMode.opi.build.flash_freq=80m
esp32s3.menu.FlashMode.opi.build.flash_type=opi
```

Which arduino settings are working with your PSRAM ? can you share all ?

liqijian101, 2022-06-26:
![Uploading Z5D%7]VR]``]S}@8M{}~7T6(1).png…]()

liqijian101, 2022-06-26:
[image]

luc-github, 2022-06-26:
I found this https://community.platformio.org/t/how-development-esp32-s3-wroom-2/28232/3
I am not sure it is necessary to recreate a json or just added them manually and which values are necessary
at least this one should be necessary
`board_build.memory_type = opi_opi`

luc-github, 2022-06-26:
have look to this issue too :
https://github.com/espressif/arduino-esp32/issues/6646

luc-github, 2022-06-26:
does it help ?

liqijian101, 2022-06-26:
I tried that board_build.memory_type = qspi_opi  or  board_build.memory_type = opi_opi .
but it also has errors.

luc-github, 2022-06-26:
you must add arduino as json has it 
`board_build.arduino.memory_type = opi_opi`

liqijian101, 2022-06-26:
it works. thanks very much.

board_build.arduino.memory_type = qspi_opi

luc-github, 2022-06-26:
cool

luc-github, 2022-06-26:
Does the cam works with ESP3D now ?

liqijian101, 2022-06-26:
not works.
I have checked the pin of camera and it is correct. But there are no images on webui

liqijian101, 2022-06-26:
[ESP3D][camera.cpp:40] handle_snap(): Camera stream reached
[ESP3D][camera.cpp:72] handle_snap(): Camera capture ongoing
[ESP3D][camera.cpp:75] handle_snap(): Camera capture failed
[ESP3D][camera.cpp:40] handle_snap(): Camera stream reached
[ESP3D][camera.cpp:72] handle_snap(): Camera capture ongoing
[ESP3D][camera.cpp:75] handle_snap(): Camera capture failed
[ESP3D][camera.cpp:40] handle_snap(): Camera stream reached
[ESP3D][camera.cpp:72] handle_snap(): Camera capture ongoing
[ESP3D][camera.cpp:75] handle_snap(): Camera capture failed
[ESP3D][camera.cpp:40] handle_snap(): Camera stream reached
[ESP3D][camera.cpp:72] handle_snap(): Camera capture ongoing
[ESP3D][camera.cpp:75] handle_snap(): Camera capture failed
[ESP3D][camera.cpp:40] handle_snap(): Camera stream reached
[ESP3D][camera.cpp:72] handle_snap(): Camera capture ongoing
[ESP3D][camera.cpp:75] handle_snap(): Camera capture failed

luc-github, 2022-06-26:
did you try https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer ?

liqijian101, 2022-06-26:
it can not work.Maybe there's something wrong with my board. I'll keep checking

liqijian101, 2022-06-27:
HI  Luc.
my board can work. It is a circuit welding problem. thaks for your help.

luc-github, 2022-06-27:
恭喜恭喜