luc-github/ESP3D #783: ESP3D does not work with BTT TFTxx

alexgleb, 2022-07-04:
This is the main requirements of BTT TFT: 

General options which MUST be activated:

EEPROM_SETTINGS (in Configuration.h)
BABYSTEPPING (in Configuration_adv.h)
AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
AUTO_REPORT_POSITION (in Configuration_adv.h)
M115_GEOMETRY_REPORT (in Configuration_adv.h)
M114_DETAIL (in Configuration_adv.h)
REPORT_FAN_CHANGE (in Configuration_adv.h)

Options to support printing from onboard media:

SDSUPPORT (in Configuration.h)
LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

luc-github, 2022-07-05:
what is the error ?

alexgleb, 2022-07-05:
> I read your explanations of defects concerning upload file functionality
> and read the code, I prepared a firmware of BTT TFT32 and a firmware of
> Marlin 2.0. It includes the main requirements described in the ticket, and
> I made a conclusion the ESP3D main solution of upload file is completely
> wrong because this solution is making some conflicts between BTT TFT
> communication with Marlin 2.0. Of course, I installed esp8266 and made a
> firmware of esp3d but all this functionality is completely useless without
> upload file

this is a scope
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
esp3d is mounted in the TFT board

luc-github, 2022-07-05:
it does not answer any of my questions

alexgleb, 2022-07-05:
it does, I got the same problems of upload file, they met, though M20 and
M21 works perfectly, the main results: upload file does not work due BTT
TFT firmware communicates with Marlin.

alexgleb, 2022-07-05:
> ESP3D is the main Issue :)
> So issue is not ESP3D - why posting here ?

because it does not do the declared functionality

alexgleb, 2022-07-05:
and an explanation looks like - others did it impossible  :)

luc-github, 2022-07-05:
No, as you mentionned BTT communicate with Marlin this is Marlin/BTT issue, the protocol M28/M29 does not handle extra communication in same time -  when doing M28/M29 Marlin/TFT FW should not do anything else that why the fix I mentioned should be present, but you did not answered if it was present on your FW,

You did not answered what error is raised during upload neither - so I have no idea what error you have neither what is the possible root cause

Unlike ESP3D most of  softwares do not care if  the upload is correct or not,  they just stream the  data and the result can be hazardous because no check is done.

So I am not sure what do you expect from ESP3D to do with Marlin / TFT behavior

Anyway the M28/M29 protocol is dropped in new version of ESP3D because of such behavior not well handled by Marlin/Repetier/etc and  TFT FW  - so  there will have not fix unless you submit one

alexgleb, 2022-07-05:
As a whole, it looks nice but 1 function of upload file does not work due
to the wrong implementation, I saw the 3.0 and installed. It looks like
much better
but to implement it in the whole scope it needs a lot of time. a year or
two . To fix such behavior it needs much less time.
I got a scope of errors with the upload file, mainly the cause is some
transfer data error due to the hardware functionality, for all cases, SD
works perfectly
but ESP3D does not do it. and I saw your explanations, I'm not happy with
this way of fan work. I'll fix it for myself when I find the solution.
because the upload file function needs to be reworked. it's not a big work.
but others should know about your unfinished work to do not waste time.

luc-github, 2022-07-05:
what unfinish work?

Digifan, 2022-07-05:
Hi! All are finished. Thanks!

С уважением,
Андрей Даниленко
Тел.  095-4921655

Отправлено из приложения "Почта" для Windows 10

От: Luc
Отправлено: вторник, 5 июля 2022 г. в 18:09
Кому: luc-github/ESP3D
Копия: Subscribed
Тема: Re: [luc-github/ESP3D] ESP3D upload file is unable to work with BTT TFTxx (Issue #783)

what unfinish work?

alexgleb, 2022-07-05:
you have at least 2 defects, one at start when uploading it changes the
state of the board where ESP is mounted and another you use the same
websocket for all purposes uploading communications and so on and you
explain it as the defects on other sides :) Marlin,  BTT, others did your
defects :)

alexgleb, 2022-07-05:
I believe when a work is not finished at some side and another work will
never finished on other side, people never get working version 3.0

luc-github, 2022-07-06:
I still do not understand your explainations
1 -  ESP3D does not change of board 
2 - websocket is not used for file upload
3 - I did not says other are defect - I said M28/M29 protocol do not allows others commands when working and so if others FW still send commands during upload there is no solution for ESP3D to fix it 

That said I really do not understand what you are doing 
1 - you report an issue but you ignore the issue template
2 - you also ignored the bot message telling you to use the template
3 - you ignored my questions to understand the error
4 - when I answered you can contribute to it to fix the issue you saw -you talked about V3 and will need 1 or 2 years 
5 - you wrote ESP3D  has defects (I am sure it has several, it is definitly not bug free) but you describe issues not related to the title : you a not happy the way fan work, websocket, ESP3D 3.0 which does not support the feature because of M28/M29 reliability 

This is an open source project, I do on my free time, which means 
1 - anyone can use it
2 - anyone can contribute to it or do it's own version
3 - no one have to pay for it 

I really do not understand why you are so unhappy about this, you are not obliged to use it, you did not pay for it, but you are welcome to contribute  

So I think you are not looking for help but just trolling

I close issue as it goes nowhere

closed 2022-07-06 by luc-github