luc-github/ESP3D #784: [BUG]ESP3D Client Station Connection error MKS Tinybee

3DSmitty, 2022-07-10:
**Describe the bug**

I am not sure if this a ESP3D, Marlin, or MKS Tinybee issue. If you are not aware the MKS Tinybee has built in ESP32-WROOM-32U as the main MCU that controls the whole board. ESP3D works great in AP Mode on the Tinybee. When I switch it to Client Station I can connect to the ESP3D webui and control the printer. After about 30 seconds it displays "Connection error" on the ESP3D console and the printer no longer responds. Then if I refresh the webpage it is not able to connect. I checked the router (openwrt) and it shows the printer is still connected even though I am not able to ping it. I can still control the printer from the LCD (mks mini12864). What could be causing this issue and how do I fix it?

**To Reproduce**

Switch from AP Mode to Client Station Mode (DHCP)

**Expected behavior**

Expect it to work without Connection Errors.

**Screenshots**

[image]
[image]
[image]
[image]
[image]

**ESP3D Firmware:**

- ESP3D FW version:

M114 P0 [ESP420]PLAIN
Chip ID: 57425
CPU Frequency: 240Mhz
CPU Temperature: 68.9C
Free memory: 125.03 KB
SDK: v3.2.3-14-gd3e562907
Flash Size: 8.00 MB
Available Size for update: 3.19 MB
Available Size for SPIFFS: 1.43 MB
Baud rate: 115200
Sleep mode: None
Web port: 80
Hostname: MP10
Active Mode: STA (**:**:**:**:**:**)
Connected to: ******
Signal: 94%
Phy Mode: 11n
Channel: 1
IP Mode: DHCP
IP: 192.168.***.***
Gateway: 192.168.***.1
Mask: 255.255.255.0
DNS: 192.168.***.1
Disabled Mode: AP (**:**:**:**:**:**)
FW version: Marlin 2.1 (ESP3D:1.0.11)

- ESP3D-WebUI Version: 2.1b77
- Wifi mode: STA
- Flash method: PlatformIO
- Any change done in FW:

**Target Firmware:**

- Name: Marlin
- Version: 2.1

M115
Ok
FIRMWARE_NAME:Marlin 2.1 (Jul  9 2022 23:44:48) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:MKS TinyBee EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:HOST_ACTION_COMMANDS:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:MULTI_VOLUME:0
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:LFN_WRITE:0
Cap:CUSTOM_FIRMWARE_UPLOAD:0
Cap:EXTENDED_M20:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
Cap:CONFIG_EXPORT:0
ok

**Board used (please complete the following information):**

- MCU: ESP32-WROOM-32U, Flash: 8192K, RAM: 520K, the frequency is: 240MHz
- Name: MKS Tinybee
- Flash size: 8M

**Additional context**

luc-github, 2022-07-10:
did you read this https://github.com/luc-github/ESP3DLib/issues/44
it looks like your issue

3DSmitty, 2022-07-10:
@luc-github Thank you so much! That worked, I have been tearing my hair out for days! Why is the "^" in Marlin if it doesn't currently support ESP32SSDP version 1.2.0. Is this a bug that needs to be submitted to Marlin?

luc-github, 2022-07-10:
not sure when it was added I discovered recently when doing some test on latest marlin

3DSmitty, 2022-07-10:
https://github.com/MarlinFirmware/Marlin/issues/24479

luc-github, 2022-07-10:
actually the issue is due to esp32 core  version/ platformio version used in Marlin, it is pretty old, here an updated version with esp3dlib V2 if you want to give a try https://github.com/luc-github/Marlin

the webui 3.0 is very different FYI

3DSmitty, 2022-07-10:
Thanks I will give it a go!

luc-github, 2022-07-17:
Can I close issue ?

3DSmitty, 2022-07-17:
Yes, sorry the fix has been applied in Marlin. Just means when ESP3D 3.0 becomes official that will have to be updated. Do you have any timeline plans to submit a PR to Marlin for 3.0?

luc-github, 2022-07-17:
Thank you, ESP3DLib 3.0 will need more than a carrot to be modified in Marlin, so it is fine  ^_^ 

>Do you have any timeline plans to submit a PR to Marlin for 3.0

So far no - I am also on process to add ESP32-S3 support but some part are still missing like I2S which is a bit out of my knowledge so it need time, to learn it or someone to help on this

closed 2022-07-17 by luc-github

3DSmitty, 2022-07-17:
Haha, ok. I know basic concept of i2c but not enough to program.