luc-github/ESP3D #785: [Feature Request]ESP700 does not handle comments

luc-github, 2022-07-16:
`;` is usually used to comment GCODE files
it can be at start of line or at end of GCODE command 
```
;change layer
G1 Z1.2 F300 ;new layer
```
Currently [ESP700] using gcode  stream host processing task just read line and send it, it would be good to save some data bandwidth, to clear the comments as I do not think they are used by FW (TBC)

sschimansky, 2022-07-16:
Sehr geehrte Damen und Herren,

vielen Dank für Ihre E-Mail.

Ich bin bis zum 2. August 2022 nicht im Büro und habe nur beschränkten Zugriff auf meine E-Mails. 

Bitte wenden Sie sich in dringenden Fällen an:

Herrn Matthias Christoph
Tel.: 05321-7591-73 60
***@***.***

Viele Grüße aus Goslar,
Sebastian Schimansky
Leiter IT-Abteilung, Entwicklung

Tel.: +49 (0) 5321-7591-62
Fax: +49 (0) 5321-7591-69

TMA TourismusMarketing Agentur GmbH | Spitalstraße 1 | D-38640 Goslar | www.tourismus-agentur.de |Sitz der Gesellschaft: Goslar | Geschäftsführer: Ralf Hummel | Ust-IdNr. DE 233056626 | HRB 111371 - AG Braunschweig

`;` is usually used to comment GCODE files
it can be at start of line or at end of GCODE command 
```
;change layer
G1 Z1.2 F300 ;new layer
```
Currently [ESP700] using gcode  stream host processing task just read line and send it, it would be good to save some data bandwidth, to clear the comments as I do not think they are used by FW (TBC)

luc-github, 2022-07-23:
Actually it does https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/gcode_host/gcode_host.cpp#L296-L307
I do not know why I forget it orz...

closed 2022-07-23 by luc-github