luc-github/ESP3D #793: problems loading ESP32 firmware from a BTT SKR 3 board.

dstrohl, 2022-07-26:
**Describe the bug**
I suspect this is not actually a bug and is instead something that I did wrong...  but here you go.

I am trying to install a ESP32 Board on a BTT SKR3 Motherboard and I am not seeing the ESP3D wifi network.  I suspect that the firmware is not installed (and probably not correctly compiled).

After attempting to install the firmware on the ESP, I am not seeing a wifi network with the name of ESP3D. I AM seeing one called ESP_xxxxxx (with the xxxxxx being the last 6 of the MAC address on the board) when I connect to that network, (no password), it provides an address in the 192.168.4.x range, and the .1 host does not accept any connections. (since this popped up immediately after the board was brought up, and goes away when it's powered down, and since the xxxx matches the MAC address, I'm pretty sure it's being generated by the card. but I don't know what it is since that's not mentioned in the docs anywhere.

**To Reproduce**

I installed a ESP32 board on a BTT SKR 3 EZ motherboard per the docs,
I configured Marlin serial port 3 per the docs and installed that version of marlin firmware on the SKR board.
I downloaded the current ESP firmware from the wiki link and compiled it in Platform IO. *no changes in any config files.
I copied the firmware.bin file from the esp8266 directory to an sd card and named it esp3d.bin, then inserted that in the board's SD card slot and reset the board. (the rename was per the SKR docs which said to do that).
the red status lite blinked for a few minutes on both the SKR and ESP boards... then the status lite went out.
according to the docs, the file should be renamed to esp3d.cur afterwards, but the .bin file remained.

The instructions seem to imply that I need to install the firmware before installing the card on the board... but all of the examples that I've found seem to be pointing to a different hardware (ESP01 or ESP07?) with a totally different header.

**Expected behavior**

**ESP3D Firmware:**

I downloaded the repo version 2.1.x (at least that's what it says at the top of the directory) as of a few days ago.
Compiled on Visual Code using PlatformIO
No changes to any of the firmware of configuration values, 
To compile, in the PlatformIO left nav, I clicked on project tasks / default / genera / build all.
the file in the esp8266 directory was the only firmware.bin that I found, so that's what I used.

**Target Firmware:**

Marlin v 2.1

**Board used (please complete the following information):**

BTT SKR v3.0 EZ
ESP32-WROOM-32U
4M Flash (I think)

**Additional context**

I am totally new to all of this, so I assume that I messed up something somewhere... so I apoligize in advance for whatever dump thing  I missed!

Wiring, decoded stack, etc...-->

luc-github, 2022-07-26:
you compiled for esp8266 which is not esp32 so the generated FW is not for esp32 and so the flashing the Esp32 board with ESP8266 FW failed
you need to click on the plaformio icon to select the esp32 dev
[image]

as you have  a ESP32-WROOM-32U I suggest you to flash it directly from platformio so you can see the flashing progress and status

dstrohl, 2022-07-27:
Thanks, is there an instruction guide to how to connect it to the pc to flash it?

luc-github, 2022-07-27:
just use usb cable (ESP board - PC) and the process I gave you

dstrohl, 2022-07-27:
Is there a diagram for which wires of the usb cable connect to which pins?  And, im assuming it’s direct wire connection. No special components needed in the middle?  (I did look around the wiki and faqs but didn’t see it covered anywhere)

dstrohl, 2022-07-27:
Thanks for your help by the way!!!

luc-github, 2022-07-27:
wait are you using just the MCU as it is ? no voltage regulator ? this won't be stable at all

luc-github, 2022-07-27:
have a look to this board it may help you: https://github.com/Xinyuan-LilyGO/T-01C3
but this is for esp32-C3 not esp32 standard, serial pins are differents

luc-github, 2022-08-14:
I guess issue can be closed now

closed 2022-08-14 by luc-github