luc-github/ESP3D #807: Won't detect SD card

jwinfieldnz, 2022-09-02:
SD card is not detected

The ESP32CAM SD card doesn't seem to be detected too often. I have used a few cards from 8 to 32GB FAT and exFAT format but doesn't seem to detect the card at all. Maybe I have the SD card settings wrong, It does work on some occasions. 

Arduino debug provides 
```
E (84334) sdmmc_sd: sdmmc_check_scr: send_scr returned 0xffffffff
E (84334) vfs_fat_sdmmc: sdmmc_card_init failed (0xffffffff).
M21
```
Maybe a faulty card reader?

[image]

[image]

-Marlin
- MCU: ESP32CAM

luc-github, 2022-09-02:
yes looks hardware/driver issue, 
can you share your cofiguration.h
Did you do any change in FW? 
what version of esp32 core do you use in arduino?

jwinfieldnz, 2022-09-02:
Sorry. You can close this. I just tried a camera sketch and fails with that also.

luc-github, 2022-09-02:
try older core/ latest esp32 core git., there was some issue with sdmmc driver on some version

liqijian101, 2022-09-03:
My esp32s3 has the same problem. Some cards can not be detected, and another 16g card is identified as 2G.

Can you recommend a suitable version

luc-github, 2022-09-03:
normaly latest git should be ok as well as 2.0.4 as well as version  currently used in esp3d V3 platformio

be sure also your card is formated using https://www.sdcard.org/downloads/formatter/

on my esp32cam the 16GB is well detected :
[image]

luc-github, 2022-09-07:
I close issue then

closed 2022-09-07 by luc-github