luc-github/ESP3D #820: [FEATURE REQUEST] Print progress status

Caian, 2022-10-09:
**Is your feature request related to a problem? Please describe.**

To check the printing progress it is necessary to click the progress button on the SD pane, wait for the M27 response and manually calculate a percentage from the byte numbers returned in the response.

**Describe the solution you'd like**

A periodical update of the progress (maybe with `M27 SXX`?) with an automatically calculated percentage or a progress bar. It would be nice as well to have an elapsed time counter and an estimation of the time left.

**Describe alternatives you've considered**

A sufficient approach would be to just parse the M27 response on the command log and print a percentage along with it. I believe it would be a front-end-only modification.

**Additional context**

None

luc-github, 2022-10-09:
What version are you  using ?

Caian, 2022-10-10:
I'm using ESP3DLib 1.0.11 as part of the firmware bundle that is shipped with MKS TinyBee, including the gz for the index page. Can I just upload a newer interface?

luc-github, 2022-10-10:
No to use ESP3D-WebUI  3.0 : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0
you need to use ESP3DLib V3 https://github.com/luc-github/ESP3DLib/tree/3.0, which only available in this version of Marlin as still in devt : https://github.com/luc-github/Marlin/tree/ESP3D-V3-2.1.x

Caian, 2022-10-10:
Could you backport at least some form of percentage string to the 2x branch?

luc-github, 2022-10-10:
no plan so far , I am already busy with v3 
there are old pr for progression but they were not integrated because they only worked on Marlin and UI was broken on some configuration

why not try the V3 ?

Caian, 2022-10-11:
I'll see if I can get the latest development branch to compile for the MKS TinyBee so I can test V3...

luc-github, 2022-10-11:
use the Marlin link I gave above - it is already setup for tinybee

closed 2022-10-11 by luc-github