luc-github/ESP3D #823: [Question] Switching between SPI and SDIO

pagabee, 2022-10-19:
Hello

I read in a [comment](https://github.com/luc-github/ESP3D/issues/794#issuecomment-1196329693), it is recommended to use a switch to share the SD card.
It is possible to switch between the SPI and the 4 bits SDIO protocol?
I mean the 3D printer board would communicate with the SD card via SPI and the ESP3D (ESP32) via SDIO.
The upload speed would be higher with SDIO but I don’t know if this kind of sharing would work or not.

luc-github, 2022-10-19:
what is your SD sharing solution ? does your printer sd reader is supporting SDIO communications ?

pagabee, 2022-10-19:
I have an MKS Robin nano V3 board with MKS wifi, but I’m not very satisfied with the MKS Marlin firmware, because it has a lot of small bugs.
The upload speed via MKS wifi is not so bad ~77kBs-100kBs, but it takes a many time to send a larger (~25MB) gcode file.
Therefore I want to find a solution for faster file upload.

The board does not support the SDIO.
My idea is shared SD with an additional board. SDIO communication with ESP32, SPI communication with the printer board.
As switching component I would use a FST3257, FST3245 or similar IC.

luc-github, 2022-10-19:
We are talking about an esp8266 chip do not expect miracle, you may increase a little bit the upload but I do not think so much - did you tried what the upload speed would be on SDIO card using 4 bits on esp8266 using arduino sample codde https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/FSBrowser ?

From esp3d point of view it should be transparent because ESP3D will only use SDCard in sdio   when it control sdcard reader, so all should depend about your hardware switch, your SDCard reader, your sd card quality and your wiring.   
But currently there is not SDIO support for esp8266...

pagabee, 2022-10-20:
I checked the upload speed on a ESP8266 board, it is ~155kBs.